kernelkit/infix #125: Hints when using wrong name of (physical) interface

jovatn, 2023-08-29:
Using the wrong name of an Ethernet interface (here 'eth0' instead of 'e0') can become confusing for a CLI user.
- While configuring, everything looks normal
- But when executing "leave", there is a general error message *"Error: Unable to commit candidate to running (9)"*, which left me confused.

Perhaps the error message could have been more informative, e.g., *"Error: Unable to commit candidate to running (9). Unable to create interface eth0."*

Or there could have been an indication already when trying to edit/create the non-existing interface. But I guess that is too complex.

The example below illustrates the issue.
- first I edit/create a bridge interface. That works fine. The name "br0" implicitly defines this as a bridge interface.
-  then I try to edit interface eth0. But in reality, I try to create eth0. It is unclear what the system thinks about its "type". Anyhow, everything goes well until I do leave. Then the error message occurs
- Removing the interface (eth0) and instead editing e0 works fine

```
Infix by KernelKit (console)
infix-00-00-00 login: root
root@infix-00-00-00:~$ cli
root@infix-00-00-00:/> configure 
root@infix-00-00-00:/config/> edit interfaces interface br0
root@infix-00-00-00:/config/interfaces/interface/br0/> set ipv4 address 10.0.1.1 prefix-length 24
root@infix-00-00-00:/config/interfaces/interface/br0/> diff
+interfaces {
+  interface br0 {
+    type bridge;
+    ipv4 {
+      address 10.0.1.1 {
+        prefix-length 24;
+      }
+    }
+  }
+}
root@infix-00-00-00:/config/interfaces/interface/br0/> leave
root@infix-00-00-00:/>                                                           <=== Success

root@infix-00-00-00:/> configure 
root@infix-00-00-00:/config/> edit interfaces interface eth0
root@infix-00-00-00:/config/interfaces/interface/eth0/> set ipv4 address 10.0.2.1 prefix-length 24
root@infix-00-00-00:/config/interfaces/interface/eth0/> diff
+interfaces {
+  interface eth0 {
+    ipv4 {
+      address 10.0.2.1 {
+        prefix-length 24;
+      }
+    }
+  }
+}
root@infix-00-00-00:/config/interfaces/interface/eth0/> leave
Error: Unable to commit candidate to running (9)
root@infix-00-00-00:/config/interfaces/interface/eth0/>                         <=== Failure

root@infix-00-00-00:/config/interfaces/interface/eth0/> up
root@infix-00-00-00:/config/interfaces/> show
interface br0 {
  type bridge;
  ipv4 address 10.0.1.1 prefix-length 24;
}
interface e0 {                                                                  <=== e0!
  type ethernet;
  ipv6 enabled true;
}
interface eth0 ipv4 address 10.0.2.1 prefix-length 24;                          <=== eth0 (=> e0)
interface lo {
  type loopback;
  ipv4 address 127.0.0.1 prefix-length 8;
  ipv6 address ::1 prefix-length 128;
}
root@infix-00-00-00:/config/interfaces/> no interface eth0
root@infix-00-00-00:/config/interfaces/> edit interface e0
root@infix-00-00-00:/config/interfaces/interface/e0/> set ipv4 address 10.0.2.1 prefix-length 24 
root@infix-00-00-00:/config/interfaces/interface/e0/> diff
interfaces {
  interface e0 {
+    ipv4 {
+      address 10.0.2.1 {
+        prefix-length 24;
+      }
+    }
  }
}
root@infix-00-00-00:/config/interfaces/interface/e0/> leave
root@infix-00-00-00:/>                                                           <=== Success
```

troglobit, 2023-08-30:
Thank you for taking the time to report this.  I suspect this feature was lost when upgrading klish to the latest version.

See also issue #111

troglobit, 2023-11-07:
This should do the trick, right? ;-)

```
~/src/infix/output/target(klish-upgrade)$ ssh -l admin fe80::ff:fe00:0%virbr0
admin@fe80::ff:fe00:0%virbr0's password: 
admin@infix-00-00-00:/> configure 
admin@infix-00-00-00:/config/> edit interfaces interface eth0
admin@infix-00-00-00:/config/interfaces/interface/eth0/> leave
Error: Mandatory node "type" instance does not exist. (Data location "/ietf-interfaces:interfaces/interface[name='eth0']".)
Failed committing candidate to running: Validation failed
admin@infix-00-00-00:/config/interfaces/interface/eth0/> set type ethernet 
admin@infix-00-00-00:/config/interfaces/interface/eth0/> leave
Error: Cannot create fixed Ethernet interface eth0, wrong type or name.
       Please check your changes, try 'diff' and 'do show interfaces'.
admin@infix-00-00-00:/config/interfaces/interface/eth0/> diff
+interfaces {
+  interface eth0 {
+    type ethernet;
+  }
+}
admin@infix-00-00-00:/config/interfaces/interface/eth0/> do show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
e0              ethernet   UP          02:00:00:00:00:00                        
                ipv6                   2001:db8:0:1:0:ff:fe00:0/64 (link-layer)
                ipv6                   fe80::ff:fe00:0/64 (link-layer)
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv6                   ::1/128 (static)
```

troglobit, 2023-11-07:
See #197 for the PR to fix this.

commit 500ae3f8 (2023-11-07): Fix #125: improve feedback on invalid configure input

This is an attempt at improving the error reporting from klish-plugin-infix,
or more specifically commit@infix.  Previously none of the sysrepo errors were
shown, now all the latest errors, as well as a few new ones specific to
interfaces have been added.

Example (eth0 does not exist):

    admin@infix-00-00-00:/config/> edit interfaces interface eth0
    admin@infix-00-00-00:/config/interfaces/interface/eth0/> leave
    Error: Mandatory node "type" instance does not exist. (Data location "/ietf-interfaces:interfaces/interface[name='eth0' [... cut]
closed 2023-11-07 by troglobit