kernelkit/infix #138: Not possible to have static IP and DHCP at the same time

jovatn, 2023-09-15:
Not sure if this is a bug or feature.

Setting a static IPv4 (and link-local autoconf), and later enabling assignment via DHCP seems to vipe the static address?!
The link-local is kept, though.

```
root@infix-00-00-00:/> show running-config 
interfaces {
  interface eth0 {
    type ethernet;
    ipv4 {
      address 10.0.1.1 prefix-length 24;
      autoconf enabled true;
    }
    ipv6 enabled true;
  }
  interface lo {
    type loopback;
    ipv4 address 127.0.0.1 prefix-length 8;
    ipv6 address ::1 prefix-length 128;
  }
}
...
}
dhcp-client {
  enabled true;
  client-if eth0
}
root@infix-00-00-00:/> show interfaces 
INTERFACE       STATE          PROTOCOL/ADDRESS          SOURCE                 
lo              up             00:00:00:00:00:00         unknown
                               127.0.0.1/8

eth0            up             02:00:00:00:00:00         unknown
                               169.254.1.3/16
                               10.1.2.100/24

root@infix-00-00-00:/>
```

troglobit, 2023-09-15:
Definite bug. But I guess a feature of the default udhcpc script shipped by BusyBox. 

Thank you for reminding me! We should target a fix for this in the next release.

troglobit, 2023-09-21:
Fixed by @rical in f75053c931273006d6d34abb34d9b5563e65ad50 for PR #135.

closed 2023-09-21 by troglobit