kernelkit/infix #173: Not possible to scp from infix

mattiaswal, 2023-10-18:
From my PC I try to initiate a SCP from infix.
lazzer@tollan ~/Documents/addiva/infix/infix-x86_64 (main)$ scp admin@192.168.100.1:/tmp/data3 /tmp/
admin@192.168.100.1's password: 
scp -f /tmp/data3
Error:
Illegal command

troglobit, 2023-10-18:
Default login shell for admin user is clish, so that won't work currently.  Needs a magic redirect or something to a shell.

terefang, 2023-11-03:
why not have "proftpd" in ssh/sftp mode runnig on port 2222 ?

troglobit, 2023-11-23:
That's also a possibility, but I think we're more likely to go for sftp using openssh subsystem.

terefang, 2023-11-28:
that might also cause problems because i know at least one client (WinScp maybe others), that will fallback to scp based filetransfers because of performance reasons

wkz, 2024-01-17:
This will be fixed by #266

closed 2024-01-17 by wkz

troglobit, 2024-01-17:
Nice take on this one, @wkz, very appreciated!