kernelkit/infix #177: BUG: possible to set bridge-port br0 on bridge br0

mattiaswal, 2023-10-26:
root@infix-00-00-00:/config/interfaces/interface/br0/> set bridge-port bridge br0
root@infix-00-00-00:/config/interfaces/interface/br0/> leave 
find: /run/net/3/dag/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0/br0: Too many levels of symbolic links

mattiaswal, 2023-10-26:
Tried to fix without success:
diff --git a/src/confd/yang/infix-if-bridge@2023-08-21.yang b/src/confd/yang/infix-if-bridge@2023-08-21.yang
index e56e3b3..7588d85 100644
--- a/src/confd/yang/infix-if-bridge@2023-08-21.yang
+++ b/src/confd/yang/infix-if-bridge@2023-08-21.yang
@@ -148,6 +148,7 @@ submodule infix-if-bridge {
 
         leaf bridge {
           type if:interface-ref;
+         when "!derived-from-or-self(../../if:type,'ianaift:bridge')";
           must "deref(.)/../bridge" {
             error-message "Must refer to a bridge interface.";
           }

troglobit, 2023-10-26:
This and the nasty klish bugs we can start working on after the two big customer releases in November.

commit 22c8e9ad (2023-12-05): Fix #177: ensure bridge is not bridge port to itself

admin@infix-00-00-00:/> configure
admin@infix-00-00-00:/config/> edit interface br0
admin@infix-00-00-00:/config/interface/br0/> set bridge-port bridge br0
admin@infix-00-00-00:/config/interface/br0/> leave
Error: Must refer to a bridge interface (and not itself). (Data location "/ietf-interfaces:interfaces/interface[name='br0']/infix-interfaces:bridge-port/bridge".)
Failed committing candidate to running: Validation failed
commit 415632f3 (2023-12-05): Fix #177: ensure bridge is not bridge port to itself

admin@infix-00-00-00:/> configure
admin@infix-00-00-00:/config/> edit interface br0
admin@infix-00-00-00:/config/interface/br0/> set bridge-port bridge br0
admin@infix-00-00-00:/config/interface/br0/> leave
Error: Must refer to a bridge interface (and not itself). (Data location "/ietf-interfaces:interfaces/interface[name='br0']/infix-interfaces:bridge-port/bridge".)
Failed committing candidate to running: Validation failed
closed 2023-12-05 by mattiaswal