kernelkit/infix #240: Address not removed from device but removed in configuration

mattiaswal, 2023-11-30:
Running config: [running.json](https://github.com/kernelkit/infix/files/13514033/running.json)
<pre>
admin@infix-06-01-00:/config/> edit interface lo 
admin@infix-06-01-00:/config/interface/lo/> set ipv4 
admin@infix-06-01-00:/config/interface/lo/> set ipv4 address 192.168.200.1 prefix-length 32
admin@infix-06-01-00:/config/interface/lo/> leave 
admin@infix-06-01-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv4                   192.168.100.1/32 (static)
                ipv4                   192.168.200.1/32 (static)
                ipv6                   ::1/128 (other)
x1              ethernet   DOWN        e2:02:a5:06:01:01                        
x10             ethernet   UP          e2:02:a5:06:01:0a                        
                ipv6                   fe80::e002:a5ff:fe06:10a/64 (link-layer)
x2              ethernet   DOWN        e2:02:a5:06:01:02                        
x3              ethernet   DOWN        e2:02:a5:06:01:03                        
x4              ethernet   UP          e2:02:a5:06:01:04                        
                ipv4                   192.168.50.1/24 (static)
                ipv6                   fe80::e002:a5ff:fe06:104/64 (link-layer)
x5              ethernet   UP          e2:02:a5:06:01:05                        
                ipv6                   fe80::e002:a5ff:fe06:105/64 (link-layer)
x6              ethernet   DOWN        e2:02:a5:06:01:06                        
x7              ethernet   DOWN        e2:02:a5:06:01:07                        
x8              ethernet   UP          e2:02:a5:06:01:08                        
                ipv4                   192.168.10.1/24 (static)
                ipv6                   fe80::e002:a5ff:fe06:108/64 (link-layer)
x9              ethernet   DOWN        e2:02:a5:06:01:09                        
admin@infix-06-01-00:/> 
</pre>
<pre>
admin@infix-06-01-00:/> copy factory-config running-config 
</pre>
Now the 192.168.200.1/32 is still there
<pre>
admin@infix-06-01-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv4                   192.168.200.1/32 (static)
                ipv6                   ::1/128 (other)
x1              ethernet   DOWN        e2:02:a5:06:01:01                        
x10             ethernet   UP          e2:02:a5:06:01:0a                        
                ipv6                   fe80::e002:a5ff:fe06:10a/64 (link-layer)
x2              ethernet   DOWN        e2:02:a5:06:01:02                        
x3              ethernet   DOWN        e2:02:a5:06:01:03                        
x4              ethernet   UP          e2:02:a5:06:01:04                        
                ipv6                   fe80::e002:a5ff:fe06:104/64 (link-layer)
x5              ethernet   UP          e2:02:a5:06:01:05                        
                ipv6                   fe80::e002:a5ff:fe06:105/64 (link-layer)
x6              ethernet   DOWN        e2:02:a5:06:01:06                        
x7              ethernet   DOWN        e2:02:a5:06:01:07                        
x8              ethernet   UP          e2:02:a5:06:01:08                        
                ipv6                   fe80::e002:a5ff:fe06:108/64 (link-layer)
x9              ethernet   DOWN        e2:02:a5:06:01:09       
</pre>

mattiaswal, 2023-12-01:
I have not actually tested it on other interfaces, but even if it is only on loopback this is bad for the upcomming OSPF feature.

troglobit, 2023-12-01:
> I have not actually tested it on other interfaces, but even if it is only on loopback this is bad for the upcomming OSPF feature.

Of course, I was just attempting to see if we could narrow the search a bit.  I'll probably have a look at it during the weekend :)

troglobit, 2023-12-03:
I think I'll need your stress test or something to reproduce this. Spent a good while now trying to reproduce without any success, unfortunately.

mattiaswal, 2023-12-21:
Can not reproduce on virtual either, but on hardware it was very reproduceable. We need to get to the bottom of this.

troglobit, 2023-12-26:
> Can not reproduce on virtual either, but on hardware it was very reproduceable. We need to get to the bottom of this.

Agreed.  When we get back from the holidays we can team up to reproduce it reliably.  My guess is still that there's something fishy with the exceptions for loopback in `ietf-interfaces.c`.

mattiaswal, 2024-01-11:
Unable to reproduce on hardware as well, closing.

closed 2024-01-11 by mattiaswal