kernelkit/infix #259: Failed to copy factory-config to startup-config

ahmkar94, 2024-01-10:
The commands:
 `copy factory-config startup-config`
 `copy running-config startup-config` 
 fail to copy to `startup-config.cfg` after reboot.

troglobit, 2024-01-23:
Just noticed a similar issue on the NanoPi R2S, tab completing `copy running-config sta<TAB>` did not work. Typing it in manually though did work, however.

Both `copy` and tab completion are part of `src/klish-plugin-infix`.  The tab completion function, `infix_datastore()` as well as logs to stderr, which I believe is wrong.  klishd parses and mangles the output from these callbacks, so we should probably rewrite the logging functions to use `syslog()` instead as an attempt to get to the bottom of this issue.

troglobit, 2024-02-04:
- Original bug report reproduced
 - Root cause found

Fix pending on `container-basic` branch.

closed 2024-02-25 by troglobit