kernelkit/infix #264: Build for rpi4 and nanopi r2s

supapavel, 2024-01-17:
Hu, do you know how to configure infix for rpi4 and nanopi?

troglobit, 2024-01-17:
Infix is built on top of Buildroot, so all Buildroot defconfigs work also in Infix.  But for the NETCONF features (like confd & statd) to be enabled, specific RPi and R2S defconfigs need to be ported to Infix.

It so happens I have just got myself an R2S, and was planning to look at it for Infix this coming weekend.  If you like to help out testing or contributing you are most welcome!

There are basically two paths:

  1. Integrate support for R2S (and RPi) in the `aarch64_defconfig`, meaning add missing drivers to the Linux .config in `board/aarch64/`, and board specific packages to `aarch64_defconfig`
  2. Add separate `r2s_defconfig` (and similar for the RPi's) and `board/r2s/` (or similar) to hold the Linux .config and other files, inheriting from the `board/aarch64/` post-build and post-image scripts.

I think (2) is the most attractive one to start with, take one board at a time, and then integrate them into the generic aarch64 depending on demand.

Customers we support on custom boards right now use the `aarch64_defconfig` as a reference build, so it's targeted at our reference and customer boards.  It will receive the most testing and any other defconfig will need more help from the community or company sponsor.

Hope this answers your question.

minexn, 2024-01-17:
@troglobit I have several R2S units sitting idle that I used for VxLAN testing. I would be happy to do some testing with R2S.

minexn, 2024-01-17:
We are testing infix on a phyCORE-AM64x and want to contribute the code back to infix, so it would be good to see if we can integrate it into the aarch64_defconfig or a separate am64x_defconfig.

wkz, 2024-01-17:
> We are testing infix on a phyCORE-AM64x and want to contribute the code back to infix, so it would be good to see if we can integrate it into the aarch64_defconfig or a separate am64x_defconfig.

Out of curiosity, how is the handover between the bootloader and Linux done, are you using the standard image format with our U-Boot scripts or some custom glue?

troglobit, 2024-01-17:
> @troglobit I have several R2S units sitting idle that I used for VxLAN testing. I would be happy to do some testing with R2S.

Awesome, I'll let you know by the end of the weekend, hopefully!

troglobit, 2024-01-17:
> We are testing infix on a phyCORE-AM64x and want to contribute the code back to infix, so it would be good to see if we can integrate it into the aarch64_defconfig or a separate am64x_defconfig.

To begin with I think we should start with a separate `am64x_defconfig`, add it to all GitHub actions, and then merge it when it's stabilized.  That's what I'll do with the R2S defconfig too.  Hope that's OK?

(The RPi builds will come a bit later, I'm hoping to use them as internal training for our junior devs.  It'll be at least one separate build here too, that can be merged to the generic `aarch64_defconfig` later.)

troglobit, 2024-01-17:
This is more of a discussion, converting ...