kernelkit/infix #274: NanoPi R2S: traffic LEDs on eth1 (LAN) not working

troglobit, 2024-01-23:
While porting to Infix I'm sure I had the LEDs on both ports working fine at some point.  Went from Linux 5.16, and while trying out various WiFi USB plugs, I went as high as 6.7.1 before landing on 6.5.11, which Infix is currently is on.  So current `linux_defconfig` works on that newer kernels, but we have some patches that don't apply, so no way of testing before relocating them to a subdirectory.

From discussion in https://github.com/kernelkit/infix/pull/271#issuecomment-1905381201

wkz, 2024-01-23:
Is this on the LAN and/or WAN port?

Had a quick look at the [schematic](http://wiki.friendlyelec.com/wiki/images/5/59/SCH_NanoPi_R2S_V1.0-1912.pdf): the LEDs on both ports are controlled by the respective PHY.

The WAN port's PHY driver does not have any support for controlling the chip's LEDs, so it ought to run with the default settings.

The USB-to-Ethernet controller on the LAN port, messes about with some LED registers, by the looks of it, but that code hasn't changed since 5.13 :man_shrugging: 

Are you running the original bootloader or the one you built? Sometimes vendors will configure things like LEDs in U-Boot and then trust Linux not to mess with them. Maybe you're missing some vendor patch to U-Boot?

troglobit, 2024-01-23:
You guessed it, the LAN port LEDs.  

I've used the same U-Boot as is in the Buildroot default recipe.  Dunno of any vendor specific U-Boot, the device can only boot from SD-card (no NOR flash or eMMC available onboard).

Only major difference I can think of: I did the initial bringup and testing on myLinux, which uses a much newer Buildroot base than Infix currently does.  So we can table this one until we've bumped Buildroot (and the kernel?).  Like @minexn says in #271, the port still works, so just an annoyance atm.

troglobit, 2024-08-11:
Found it, the driver was not setting up the LED register.  Fixed in [kkit-linux-6.10-y @ daba70e7](https://github.com/kernelkit/linux/commit/daba70e7782b704224396f0d86dcd91b4713e064).

Patch ported to upcoming `r2s` branch.

PR #551 (2024-08-16): Update NanoPi R2S
## Description

This PR holds massive updates of the Infix support to the NanoPi R2S, initially added in #271.  From the ChangeLog:

- Update Linux kernel to v6.10.3 and sync defconfig with aarch64
- Workaround `reboot` command "hang" on NanoPi R2S (failure to reboot) by replacing the Rockchip watchdog driver with "softdog"
- Update U-Boot to v2024.07, enable secure boot loading of images
- Rename interfaces to LAN + WAN to match case and LEDs
- Rename images to `infix-r2s$ver.ext`, not same as other aarch64
- Change rootfs to squashfs for enhanced security
- Add RAUC support [... cut]
commit b0c5d748 (2024-08-16): patches/linux: port relevant kkit-6.6.y patches to kkit-6.10.y

Only the most important bridge and usb fixes have been ported in this
commit, and a new patch for the r8153 link/traffic LEDs, ported from
the FriendlyELEC/Rockchip kernel tree.

Fixes #274
closed 2024-08-16 by wkz