kernelkit/infix #283: Setting hostname is adding quotation marks to the DHCP request.

minexn, 2024-01-29:
Manually setting the hostname in the DHCP request, and quotation marks are sent to the DHCP server.

[image]

Config

```
set dhcp-client client-if eth0
set dhcp-client client-if eth0 option hostname value infix-r2s-1
```
This can be seen in the udhcpc process.

`udhcpc -f -p /run/dhcp-eth0.pid -t 10 -T 3 -A 10 -a1000 -S -R -x hostname:"infix-r2s-1" -i eth0 -V Infix v23.11.0-121-g051658d-dirty`

troglobit, 2024-01-29:
Nice catch!  I'm guessing -x "hostname:infix-r2s-1" would work better, right?

minexn, 2024-01-29:
Because the hostname doesn't contain spaces, we don't need quotation marks.

_RFCs mandate that a hostname's labels may contain only the ASCII letters 'a' through 'z' (case-insensitive), the digits '0' through '9', and the hyphen. Hostname labels cannot begin or end with a hyphen. No other symbols, punctuation characters, or blank spaces are permitted._

`udhcpc -f -p /run/dhcp-eth0.pid -t 10 -T 3 -A 10 -a1000 -S -R -x hostname:infix-r2s-1 -i eth0 -V Infix v23.11.0-121-g051658d-dirt`

closed 2024-01-30 by wkz

minexn, 2024-01-31:
Another fantastic fix for infix 👍

commit 73f9195c (2024-01-30): confd: add exception for hostname DHCP client option

No need to guard the hostname option in quotes, it must not contain
whitespace.  If user configured more than one word, use first.

Fixes #283
commit 08cec2e9 (2024-01-30): confd: add exception for hostname DHCP client option

No need to guard the hostname option in quotes, it must not contain
whitespace.  If user configured more than one word, use first.

Fixes #283