kernelkit/infix #288: RAUC support on NanoPi R2S

minexn, 2024-01-31:
I'm testing Rauc support on the NanoPi and getting the following error.

```
admin@infix-r2s-x:~$ sudo rauc status

(rauc:1323): rauc-WARNING **: 09:01:22.863: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1

Error retrieving slot status via D-Bus: error calling D-Bus method "GetSlotStatus": Failed to determine slot states: Did not find booted slot (matching '/dev/mmcblk0p1')
```

```
bash-5.2# sudo fdisk -l
Disk /dev/mmcblk0: 30 GB, 31719424000 bytes, 61952000 sectors
968000 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    2,10,9      34,172,10        32768     557055     524288  256M 83 Linux
/dev/mmcblk0p2    34,172,11   36,182,18       557056     589823      32768 16.0M 83 Linux
```

Do we need to add support in uboot?

wkz, 2024-01-31:
There are a few things one needs to get right in order to enable full RAUC support, as we've integrated it in Infix.

RAUC is governed by its config file in `/etc/rauc/system.conf`, in which we find this snippet:
```
...
[slot.rootfs.0]
device=/dev/disk/by-partlabel/primary
bootname=primary

[slot.rootfs.1]
device=/dev/disk/by-partlabel/secondary
bootname=secondary
...
```
So you need a GPT partition table on your MMC, with partition labels that match what RAUC expects.

At the moment, we use `/libexec/infix/prod/provision` to provision Infix to a blank MMC. Use it by:
1. Placing the Infix upgrade bundle (`images/infix-aarch64.pkg`) next to the netboot image (`images/rootfs.itb`) on the TFTP server
3. Netbooting the target
4. Ensure that Infix can reach the TFTP server
5. Ensure that the date is properly set, so that the upgrade bundle's signature can be verified
6. Run `/libexec/infix/prod/provision tftp://host/path/to/infix-aarch64.pkg /dev/mmcblk0`

This designed to work in combination with all of our boot scripts (and the features needed to support them) in `board/common/uboot`. So the answer to your question is: it depends.

Fundamentally, the primary and secondary partitions house regular SquashFS images (see also [boot.md](https://github.com/kernelkit/infix/blob/main/doc/boot.md)). So you _can_ boot Infix with no special requirements on U-Boot. Here's an example:
```
(board) sqfsload mmc 0:3 ${kernel_addr_r} /boot/Image
27269632 bytes read in 1558 ms (16.7 MiB/s)
(board) sqfsload mmc 0:3 ${fdt_addr_r} /boot/vendor/board.dtb
31636 bytes read in 46 ms (670.9 KiB/s)
(board) setenv bootargs "root=PARTLABEL=primary"
(board) booti ${kernel_addr_r} - ${fdt_addr_r}
```

If you want to verify image signatures, then I would look at integrating the existing scripts in the U-Boot build for the NanoPi.

troglobit, 2024-02-02:
@minexn would you be OK with my converting this to a Discussion?  I'd like to keep nuggets like this reply as part of our common "Knowledge Base".

minexn, 2024-02-02:
Please proceed