kernelkit/infix #310: cli: new VLAN support in "show interfaces" assumes VLANs

troglobit, 2024-02-23:
The new VLAN support in `show interfaces` for bridges assumes the bridge has VLAN filtering enabled.  The following bridge was set up without VLAN filtering:
```bash
admin@infix-00-00-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 vlan:1u                                  
├ e1            bridge     FORWARDING  vlan:1u pvid:1                           
├ e2            bridge     FORWARDING  vlan:1u pvid:1                           
├ e3            bridge     FORWARDING  vlan:1u pvid:1                           
└ e4            bridge     FORWARDING  vlan:1u pvid:1       
```

wkz, 2024-02-23:
I think `show interfaces` does the right thing.

https://github.com/kernelkit/infix/blob/0cb1459f43a47b951be0ef1df5c2092cb32e4cf9/src/confd/src/ietf-interfaces.c#L888-L905

Pretty sure this is the culprit. We should always supply `vlan_default_pvid=0`, otherwise the kernel will add VLAN 1 even though it is never used for filtering.

troglobit, 2024-02-26:
Can confirm, with my local patch this bug is history.  Will make a separate fix branch this time.

commit a8cd2668 (2024-02-26): confd: always create bridge with default_pvid 0

Fix #310
commit 83821def (2024-02-26): confd: always create bridge with default_pvid 0

Fix #310