kernelkit/infix #326: Refactor YANG bootstrap from online to offline

troglobit, 2024-03-04:
Currently, loading of all YANG models at runtime takes a LOT of time -- on the main quad-core Cortex A72 platform it takes up to 15 seconds! 

The reason for us not using the defaults created offline (at build time) is that sysrepo only allows initializing the factory datastore (per yang model) when loading the YANG model.

After discussions in the team we think one avenue to explore here is to patch sysrepo to unlock loading factory using `sysrepoctl` at least once at boot.  This would allow us to load device specific factory data, e.g. `admin` user's default password (unique per device).

wkz, 2024-04-02:
As part of this task we might consider making some other changes to our model:

- `/system/hostname`: Support `%`-specifiers. We should probably support at least:
   - `%i`: OS ID (from `/etc/os-release`). `infix` on vanilla builds,
   - `%m`: NIC specific part of base MAC. `c0:ff:ee`
- `/system/authentication/user/password`: Augment model to allow the literal `factory-default` or similar. This would indicate that the device's factory password, whatever it may be, should be used.

This would make it easier to support having static factory-configs, though there are still problems to solve there I think (SSH server keys come to mind).

troglobit, 2024-05-03:
Further discussions on this topic have yielded the following ideas:

 - We will need to add support for yescrypt, and similar, so updating the password hash pattern in ietf-system for `$y$` will open up for alos carrying a string like `factory-default` as well
 - `/ietf-interface:interfaces` could be extended with support for _templates_, or _profiles_

        interface e1
          type ethernet
          profile default
          ethernet speed-duplex 100 full

        interface default
          type profile
          ipv6

     The yang model could look something like:

        interface {
            leaf-ref profile;
              default "default";
        }

jovatn, 2024-05-08:
Should we make a break-down of this task?

troglobit, 2024-05-08:
Great idea, with the work of @mattiaswal we now know more about it, and the work has also shifted a bit.  It looks something like this (to be added to the issue summary if we agree):

 1. Refactor bootstrap enough to be done pre-runtime (buildtime if possible)
 2. Upgrade sysrepo, liyang, libnetconf, netopeer2, etc.
 3. Start integrating [rousette](https://github.com/CESNET/rousette) and its companion packages

There are more steps, but focus has moved slightly from speeding up boot to enabling upgrade of sysrepo to get rousette integrated so we get RESTCONF support for the proper WebUI.

jovatn, 2024-05-13:
Are these things part of this task or should we spawn or create subtasks?
- Documentation tree of supported yang models. 
- Model changes @wkz described above related to /system/hostname and /system/authentication/user/password

troglobit, 2024-05-13:
@jovatn + @mattiaswal I've made a breakdown into more tasks now.  Would've been great if there was a "depends-on" relationship for issues on GitHub ... anyway, we should have a CCB meeting and plan these.

troglobit, 2024-05-24:
Fixed in PR #460.

closed 2024-05-24 by troglobit