kernelkit/infix #328: PVID added on tagged port without PVID setting in configuration

mattiaswal, 2024-03-05:
<pre>
admin@dut2:~$ bridge vlan
port              vlan-id  
eth1              1 PVID
eth2              1 PVID Egress Untagged
br0               1

admin@dut2:/config/> show interface eth1
type etherlike;
enabled true;
ipv6
bridge-port {
  bridge br0;
}
 </pre>
still remains after copy running startup and reboot
Seen when running infamy, and ifix@ca44237b2883b56b14f7bcbb372, attaching full startup config
[startup.txt](https://github.com/kernelkit/infix/files/14495818/startup.txt)

troglobit, 2024-03-05:
Nice find!

troglobit, 2024-03-05:
> The problem only seems to exist if using VLAN id 1, changing to VLAN id 2 in the test, PVID flag was no more.

Hmm, that rings a bell. We recently changed `ietf-interfaces.c` to always create bridges with `vlan_default_pvid 0`, maybe there's a thread to pull there?

troglobit, 2024-03-06:
The root cause for this issue is that the model has "1" as default PVID.  The current code only looks at that setting to issue the PVID flag when adding the port to the VLAN.

It seems to me we need to remove the default value from the model to give the user full freedom of configuration.  E.g., tagged member in VLAN 1 but no PVID would mean untagged traffic ingressing the port would be dropped.

commit e5417ead (2024-03-07): Fix #328: drop default value for bridge-port PVID

A bridge port in a VLAN filtering bridge should be possible to configure
as "tagged only", i.e., to not assign ingressing untagged frames a VLAN
and instead drop them.
commit a0c8ae2b (2024-03-09): Fix #328: drop default value for bridge-port PVID

A bridge port in a VLAN filtering bridge should be possible to configure
as "tagged only", i.e., to not assign ingressing untagged frames a VLAN
and instead drop them.
commit 8c2da940 (2024-03-12): Fix #328: drop default value for bridge-port PVID

A bridge port in a VLAN filtering bridge should be possible to configure
as "tagged only", i.e., to not assign ingressing untagged frames a VLAN
and instead drop them.
commit 21fd1e94 (2024-03-13): Fix #328: drop default value for bridge-port PVID

A bridge port in a VLAN filtering bridge should be possible to configure
as "tagged only", i.e., to not assign ingressing untagged frames a VLAN
and instead drop them.
closed 2024-03-13 by troglobit