kernelkit/infix #357: IPv6 auto-conf for bridge interfaces gets random link identifier instead of MAC based (eui64)

jovatn, 2024-03-26:
Reproduced on latest infix (2024-03-25) but seems to have been like this all the time.

When using IPv6 autoconfiguration on a bridge interface, the fe80:: address will differ for each boot. The link identifier seems to be random rather than based on the bridge MAC (which is inherited from the first/lowest port attached)

```
admin@infix-06-11-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ipv4                   169.254.4.73/16 (random)
│               ipv4                   198.18.101.168/24 (dhcp)
│               ipv6                   fe80::206a:30ff:fe2d:5db4/64 (link-layer)
├ x1            bridge     LOWER-LAYER-DOWN
├ x2            bridge     LOWER-LAYER-DOWN
...
```
Next boot

```
admin@infix-06-11-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ipv4                   169.254.4.73/16 (random)
│               ipv4                   198.18.101.168/24 (dhcp)
│               ipv6                   fe80::9cc2:1eff:fea5:9437/64 (link-layer)
├ x1            bridge     LOWER-LAYER-DOWN
├ x2            bridge     LOWER-LAYER-DOWN
├ x3            bridge     LOWER-LAYER-DOWN
├ x4            bridge     LOWER-LAYER-DOWN
...
```

> Note that the address claims to be based on MAC address (type `link-layer` rather than `random` in listing above). 

```
admin@infix-06-11-00:~$ ip -d -j link show dev br0 | jq -r '.[].inet6_addr_gen_mode'
eui64
admin@infix-06-11-00:~$ 
```

troglobit, 2024-03-30:
Cannot reproduce (on my Qemu system).  Could you attach the startup-config you used to trigger this (censoring any customer specifics)?

Also, did this trigger from power-on (using startup-config) or did it happen at runtime, i.e., reconfiguring the device to use bridge + ipv6?  Because in the latter case we've seen issues with enabling settings post-boot that don't "take" until after a save and reboot.

jovatn, 2024-04-16:
An example with qemu using infix latest as of April 16 (built April 15 as shown below)
Configuration included, but in short.
- Default config
- Create br0, enable IPv6
- Let e0 be a bridge-port of br0
- That's it

Startup Config attached.

```
PC~:/tmp/infix-x86_64 $ ./qemu.sh 
Starting Qemu  ::  Ctrl-a x -- exit | Ctrl-a c -- toggle console/monitor
● ● ●  Infix -- a Network Operating System e06bcb6 ═════════════════════
[ OK ] Mounting filesystems from /etc/fstab
[ OK ] Restoring system clock (UTC) from RTC
[ OK ] Seeding random number generator
...
```

```
admin@infix-00-00-00:~$ cli

See the 'help' command for an introduction to the system

admin@infix-00-00-00:/> show version
Infix -- a Network Operating System e06bcb6 -- Apr 15 15:11 UTC 2024
admin@infix-00-00-00:/> 
```

Configuration:

```
admin@infix-00-00-00:/> configure 
admin@infix-00-00-00:/config/> edit interface br0
admin@infix-00-00-00:/config/interface/br0/> set ipv6 enabled 
admin@infix-00-00-00:/config/interface/br0/> show
type bridge;
ipv6 {
  enabled true;
}
admin@infix-00-00-00:/config/interface/br0/> end
admin@infix-00-00-00:/config/> set interface e0 bridge-port bridge br0
admin@infix-00-00-00:/config/> leave
admin@infix-00-00-00:/> copy running-config startup-config 
admin@infix-00-00-00:/> 
```
Show IP assignment

```
admin@infix-00-00-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ethernet   UP          02:00:00:00:00:00                        
│               ipv6                   fe80::dc7c:aeff:fef4:226/64 (link-layer)
└ e0            bridge     FORWARDING  
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv6                   ::1/128 (static)
admin@infix-00-00-00:/> 
```
reboot and login again
IPv6 changed, despite stating "link-layer" (not random)

```
admin@infix-00-00-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ethernet   UP          02:00:00:00:00:00                        
│               ipv6                   fe80::4c11:46ff:fee5:b3de/64 (link-layer)
└ e0            bridge     FORWARDING  
[... 100 lines cut ...]
        "mcast_startup_query_intvl": 3124,
        "mcast_stats_enabled": 0,
        "mcast_igmp_version": 2,
        "mcast_mld_version": 1,
        "nf_call_iptables": 0,
        "nf_call_ip6tables": 0,
        "nf_call_arptables": 0
      }
    },
    "num_tx_queues": 1,
    "num_rx_queues": 1,
    "gso_max_size": 65536,
    "gso_max_segs": 65535,
    "tso_max_size": 65536,
    "tso_max_segs": 65535,
    "gro_max_size": 65536,
    "gso_ipv4_max_size": 65536,
    "gro_ipv4_max_size": 65536
  }
]
admin@infix-00-00-00:~$ 

[startup-config.txt](https://github.com/kernelkit/infix/files/14993847/startup-config.txt)

```

troglobit, 2024-04-22:
Reproduced.  Root cause is the initial random MAC address assigned to the bridge before we add any bridge ports.  The fix is to ensure the bridge is always created with the system base mac.

For Qemu/Qeneth:
 1. Update qeneth template (dut1:eth0 -> 02:00:00:00:01:01)
 2. probe -> extract base mac or take eth0 MAC - 1
 3. confd gen bridge: if custom -> set custom, else set base mac, else random

troglobit, 2024-04-23:
Re: IPv6 addrgen mode no longer shown, apparently the iproute2 upgrade caused this.  In v6.5 they added support for "filtering stats", a kernel feature that has been available since Linux 2015, behind which the IPv6 addrgen mode is hidden.

From now on, use:

    ip -d -j -s link show dev br0 | jq -r '.[].inet6_addr_gen_mode'

The `-s` is for statistics.

troglobit, 2024-04-28:
The fix to this has been moved to a separate branch,  https://github.com/kernelkit/infix/tree/stable-bridge-mac, because it for some reason break IGMP functionality.  Pending investigation.

rical, 2024-08-21:
I had a quick look at why the test-suite fails with the fix for this in https://github.com/kernelkit/infix/tree/stable-bridge-mac applied.

The test `igmp-basic.py` fails with the message `not ok 6 - Verify that the group is no longer forwarded to the non-member`. In this scenario the test joins a multicast group on interface X and attempts to ensure traffic is now longer flooded to the non-member interface Y. After the mentioned commit is applied `nojoin_ns.must_not_receive("ip dst 224.1.1.1")` fails, indicating that tcpdump saw multicast traffic on the non-member interface.

I manually verified that bridge snooping still works and that traffic isn't flooded as the test failure suggests. This points to either a race condition in the test or some strange first-time behavior when joining a group in a pristine system. Investigation will continue.

commit 45ebaffc (2024-03-29): statd: restore MAC address for bridges in CLI "show interfaces"

Fixes #357
commit 69339c19 (2024-04-23): confd: use the system base mac as mac address on bridges

To prevent the kernel from setting a random mac address on new bridges,
before we have added any bridge ports, we create bridges using the:

  1. Custom mac address from the configuration (phys-address)
  2. System base mac from /run/system.json
  3. None, if there is no base mac address in system.json, e.g. r2s

Fixes #357
commit 68f036ad (2024-04-25): confd: use the system base mac as mac address on bridges

To prevent the kernel from setting a random mac address on new bridges,
before we have added any bridge ports, we create bridges using the:

  1. Custom mac address from the configuration (phys-address)
  2. System base mac from /run/system.json
  3. None, if there is no base mac address in system.json, e.g. r2s

Fixes #357
commit cfd95dc3 (2024-09-10): confd: use the system base mac as mac address on bridges

To prevent the kernel from setting a random mac address on new bridges,
before we have added any bridge ports, we create bridges using the:

  1. Custom mac address from the configuration (phys-address)
  2. System base mac from /run/system.json
  3. None, if there is no base mac address in system.json, e.g. r2s

Fixes #357
closed 2024-09-10 by troglobit