kernelkit/infix #358: `show interfaces` bridge formatting and content issues

jovatn, 2024-03-26:
The output of show interfaces 
- does not show the MAC address of a bridge 
- could be better formatted when lower link is down
- maybe a good idea to also write out mac of ports which are bridge-ports?

Compare infix latest 2024-02-28:
```
admin@infix-06-11-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             ethernet   UP          20:de:ad:06:11:01                        
│               ipv4                   198.18.101.168/24 (dhcp)
│               ipv6                   fe80::3cc3:2aff:fe19:9693/64 (link-layer)
├ e1            ethernet   DOWN        20:de:ad:06:11:01                        
├ e10           ethernet   UP          20:de:ad:06:11:0a                        
├ e2            ethernet   DOWN        20:de:ad:06:11:02                        
├ e3            ethernet   DOWN        20:de:ad:06:11:03                       
```

and infix latest 2024-03-25:
```
admin@infix-06-11-00:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ipv4                   169.254.4.73/16 (random)
│               ipv4                   198.18.101.168/24 (dhcp)
│               ipv6                   fe80::ec94:b1ff:fef6:865f/64 (link-layer)
├ e1            bridge     LOWER-LAYER-DOWN
├ e2            bridge     LOWER-LAYER-DOWN
├ e3            bridge     LOWER-LAYER-DOWN
├ e4            bridge     LOWER-LAYER-DOWN
├ e5            bridge     LOWER-LAYER-DOWN
├ e6            bridge     LOWER-LAYER-DOWN
```

troglobit, 2024-03-27:
Maybe `LINK-DOWN` is even better?

wkz, 2024-03-28:
I also think "Alt 1" is the way to go. I think your order is the right one, as you end up with the same stack as regular interfaces when you configure an IP address. E.g.

```
INTERFACE       PROTOCOL   STATE       DATA                                     
br0             bridge                 
│               ethernet   UP          f6:03:e8:6e:77:9f                        
│               ipv6                   fe80::deff:fead:0/64 (link-layer)                        
└ veth0b        bridge     FORWARDING  
e0              ethernet   LINK-DOWN   02:00:00:00:00:00
                ipv6                   fe80::deff:fead:1c/64 (link-layer)
```

troglobit, 2024-03-28:
I'll have a go at fixing this over the weekend

commit bfcd67fd (2024-03-30): statd: restore MAC address for bridges in CLI "show interfaces"

Fixes #358
commit 8fee83ac (2024-04-03): statd: restore MAC address for bridges in CLI "show interfaces"

Fixes #358
commit b400a6aa (2024-04-04): statd: restore MAC address for bridges in CLI "show interfaces"

Fixes #358