kernelkit/infix #419: GNS3 image built from source

minexn, 2024-04-30:
We tested GNS3 with the Infix v24.02.0 release, and everything works perfectly.

When we update the GNS3 node with the infix-x86_64-disk.img image build from the source, the console is stuck at ``Booting Infix'``

The image works when we run the .`/qemu.sh` command.

Is there anything special we must do to build a GNS3-compatible image?

troglobit, 2024-04-30:
We had a bit of a regression in the console in this latest release cycle (24.02 -> today), which "source" are you building from?  Should be fixed as of main @ c21c94f

troglobit, 2024-04-30:
@jovatn iirc, you've tested the gns3 builds on-and-off for the last couple of months, right?  I know you too mentioned console issues on aarch64 hw, but did you notice anything in gns3?

minexn, 2024-05-01:
GNS3 console seems to fail after the change from `console=ttyS0` to `console=hvc0` in the commit below,
https://github.com/kernelkit/infix/commit/c637ab4edbe16c469306e1ca51bee4fb9cf9ffd3

I believe GNS3 is looking to ttyS0.

minexn, 2024-05-01:
When I change the GRUB params at boot to the following, it works,

`set append="console=ttyS0 root=PARTLABEL=$slot $2"`

I'm not familiar with GNS3 hvc0 support

troglobit, 2024-05-01:
Reproduced and fixed on my patch queue.  (Preparing the v24.04 release right now.)

commit ab3a09b2 (2024-05-01): board/x86_64: follow-up to c637ab4, allow console also on ttyS0

Fixes #419
closed 2024-05-01 by troglobit