kernelkit/infix #432: Generate list of supported YANG models for documentation

troglobit, 2024-05-13:
As discussed in #326, after the big refactor to offline, we could generate a list of supported YANG models, their features, as well as augments/deviations.

As an example, with only a few models, here's how to create a HTML/JavaScript page of IETF System:

    pyang -f jstree ietf-system@2014-08-06.yang infix-shell-type@2023-08-21.yang \
                    infix-system@2024-04-12.yang > infix-ietf-system.html

troglobit, 2024-06-23:
Two alternatives:
 - trimmed output from tried-and-tested pyang: [output.html.gz](https://github.com/user-attachments/files/15948144/output.html.gz)
 - newcomer ytree: [ytree.html.gz](https://github.com/user-attachments/files/15948145/ytree.html.gz)

jovatn, 2024-06-24:
Wow!

commit b4b96589 (2024-06-28): board/common: extract YANG documentation for all builds

This change builds on top of the earlier work to bootstrap Infix YANG
modules at build time.  Adding a step at the end of post-build.sh to
call the yangdoc tool.  The resulting yangdoc.html is a stand-alone HTML
file of (almost) all YANG nodes.  Almost because some top-level nodes
are unused by Infix atm. and have been filtered out for readability.

Fixes #432
commit 10744c51 (2024-06-28): board/common: extract YANG documentation for all builds

This change builds on top of the earlier work to bootstrap Infix YANG
modules at build time.  Adding a step at the end of post-build.sh to
call the yangdoc tool.  The resulting yangdoc.html is a stand-alone HTML
file of (almost) all YANG nodes.  Almost because some top-level nodes
are unused by Infix atm. and have been filtered out for readability.

Fixes #432
closed 2024-06-28 by troglobit