kernelkit/infix #441: Add support for ietf-ssh-server YANG

troglobit, 2024-05-13:
Scope:

- [ ] Enable disable service
- [ ] Change listening port
- [ ] Bind to a single interface

https://github.com/YangModels/yang/blob/main/standard/ietf/DRAFT/ietf-ssh-server.yang

> **Note:** [netopeer2-server use libssh](#440) for access on port 830.  This should probably be handled by the NETCONF model and not the SSH server model.

mattiaswal, 2024-05-14:
Blocked by #434, major changes in the model in that.

troglobit, 2024-08-30:
Looks like it might be better to just extend infix-services.yang.

troglobit, 2024-11-12:
Reduced scope (enable/disable + custom port) re-targeted for v24.11.

mattiaswal, 2024-11-21:
Proposed model (infix-services):
`
module infix-services {
  yang-version 1.1;
  namespace "urn:ietf:params:xml:ns:yang:infix-services";
  prefix infix-svc;

  import ietf-inet-types {
    prefix inet;
    reference
      "RFC 6991: Common YANG Data Types";
  }
  import ietf-crypto-types {
    prefix ct;
    reference
      "RFC AAAA: YANG Data Types and Groupings for Cryptography";
  }

  import ietf-ssh-server {
    prefix ssh-srv;
  }

  import ietf-ssh-common {
    prefix ssh-common;
  }

  import ietf-tcp-server {
    prefix tcp-srv;
  }
  import ietf-keystore {
    prefix ks;
  }

  organization "KernelKit";
  contact      "kernelkit@googlegroups.com";
  description  "Infix services, generic.";

  revision 2024-06-08 {
    description "Add support for RESTCONF enable/disable as a web service.";
    reference "internal";
  }
  revision 2024-05-30 {
    description "Add support for RESTCONF enable/disable as a web service.";
    reference "internal";
  }
  revision 2024-04-08 {
    description "Initial support for web services.";
    reference "internal";
  }
  revision 2023-10-16 {
    description "Drop SSDP support, mDNS-SD is now available in Windows 10.";
    reference "internal";
  }
  revision 2023-08-22 {
    description "Initial revision, add SSDP and mDNS-SD enable/disable only.";
    reference "internal";
  }
....
  container ssh {
    leaf enabled {
      default true;
      description "Disable or enable SSH daemon";
      type boolean;
    }
    choice host-key-type {
      description
        "The type of host key being specified";
      container public-key {
        description
          "A locally-defined or referenced asymmetric key pair
           to be used for the SSH server's host key.";
        reference
          "RFC 9642: A YANG Data Model for a Keystore";
        uses ks:inline-or-keystore-asymmetric-key-grouping {
          refine "inline-or-keystore/inline/inline-definition" {
            must 'not(public-key-format) or derived-from-or-self'
              + '(public-key-format, "ct:ssh-public-key-format")';
          }
          refine "inline-or-keystore/central-keystore/"
            + "central-keystore-reference" {
            must 'not(deref(.)/../ks:public-key-format) or '
              + 'derived-from-or-self(deref(.)/../ks:public-'
              + 'key-format, "ct:ssh-public-key-format")';
          }
        }
      }
    }
    list listen {
      key "address port";
      leaf address {
        type inet:ip-address;
        description
          "The local IP address to listen on for incoming
           SSH client connections.  INADDR_ANY (0.0.0.0) or
           INADDR6_ANY (0:0:0:0:0:0:0:0 a.k.a. ::) MUST be
           used when the server is to listen on all IPv4 or
           IPv6 addresses, respectively.";
      }
      leaf port {
        type inet:port-number;
        default "22";
        description "Local port for SSH daemon to listen to.

                     The standard port for SSH is 22";
      }
    }
  }
......
}

`

troglobit, 2024-11-21:
Very nice! A few comments:

 1. Please move the description first in the leaf definitions, I've tried to be consistent with that in other models
 2. Could you post the `-f tree` view as well?

mattiaswal, 2024-11-28:
module: infix-services
  +--rw mdns
  |  +--rw enabled?   boolean
  +--rw ssh
  |  +--rw enabled?   boolean
  |  +--rw hostkey*   ks:asymmetric-key-ref
  |  +--rw listen* [name]
  |     +--rw name       string
  |     +--rw address?   inet:ip-address
  |     +--rw port?      inet:port-number
  +--rw web
     +--rw enabled?     boolean
     +--rw console
     |  +--rw enabled?   boolean
     +--rw netbrowse
     |  +--rw enabled?   boolean
     +--rw restconf
        +--rw enabled?   boolean

commit 4500254e (2024-12-04): test: Verify ssh server config

Fixes #441
commit a000e683 (2024-12-04): test: Verify ssh server config

Fixes #441
commit 65167a08 (2024-12-05): test: Verify ssh server config

Fixes #441
commit 1afe6e84 (2024-12-09): test: Verify ssh server config

Fixes #441
closed 2024-12-20 by mattiaswal