kernelkit/infix #479: containers: Not possible to disable a container

mattiaswal, 2024-06-14:
### Current Behavior

```
admin@dut4:/config/> set container alpine-nginx enabled false 
admin@dut4:/config/> leave 
Error: EV ORIGIN: SHM event "done" ID 2 processing timed out.
Please check your changes, try 'diff' and 'do show interfaces'.
admin@dut4:/config/> diff
admin@dut4:/config/> 
```
```
Jun 14 10:55:17 dut4 alpine-nginx[0]: opening fifo /run/containers/alpine-nginx.fifo
Jun 14 10:55:17 dut4 alpine-nginx[0]: entering poll loop ...
Jun 14 10:55:17 dut4 finit[1]: Stopped container:alpine-nginx[3258]
Jun 14 10:55:17 dut4 container[2658]: Container alpine-nginx has been removed.
Jun 14 10:55:19 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 2000 msec (1/10)
Jun 14 10:55:19 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:21 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 2000 msec (2/10)
Jun 14 10:55:21 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:23 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 2000 msec (3/10)
Jun 14 10:55:23 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:25 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 2000 msec (4/10)
Jun 14 10:55:25 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:27 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 2000 msec (5/10)
Jun 14 10:55:27 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:29 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 5000 msec (6/10)
Jun 14 10:55:29 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:34 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 5000 msec (7/10)
Jun 14 10:55:34 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:39 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 5000 msec (8/10)
Jun 14 10:55:39 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:44 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 5000 msec (9/10)
Jun 14 10:55:44 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:49 dut4 finit[1]: Service container:alpine-nginx[0] died, restarting in 5000 msec (10/10)
Jun 14 10:55:49 dut4 finit[1]: Calling 'container -n alpine-nginx start' ...
Jun 14 10:55:54 dut4 finit[1]: Service container:alpine-nginx keeps crashing, not restarting.
Jun 14 10:55:56 dut4 klishd-service: Stop SysRepo session for "admin"
Jun 14 10:55:56 dut4 klishd[3132]: klishd-service: Stop SysRepo session for "admin"

```
```
admin@dut4:~$ initctl 
PID   IDENT                    STATUS   RUNLEVELS     DESCRIPTION                                                                                            
1233  udevd                    running  [S-12345-789] Device event daemon (udev)
2394  dbus                     running  [S-123456789] D-Bus message bus daemon
2658  confd                    running  [S-12345----] Configuration daemon
3111  netopeer                 running  [--12345----] NETCONF server
2662  dnsmasq                  running  [S-12345----] DHCP/DNS proxy
3121  execd                    running  [---2345----] Container job runner
3125  tty:hvc0                 running  [--12345-789] Getty on hvc0
3131  iitod                    running  [S0123456789] LED daemon
3132  klishd                   running  [---2345----] CLI backend daemon
3155  mdns-alias               running  [---2345----] mDNS alias advertiser 
3134  rauc                     running  [---2345----] Software update service
3142  sshd                     running  [---2345----] OpenSSH daemon
3046  statd                    running  [S-12345----] Status daemon
2328  syslogd                  running  [S0123456789] System log daemon
2678  watchdogd                running  [S0123456789] System watchdog daemon
3148  zebra                    running  [---2345----] Zebra routing daemon
3150  mdns                     running  [---2345----] Avahi mDNS-SD daemon
3159  k8s-logger:alpine-nginx  running  [---2345----] Logger for container alpine-nginx
0     container:alpine-nginx   crashed  [---2345----] container alpine-nginx
3165  k8s-logger:curios        running  [---2345----] Logger for container curios
3221  container:curios         running  [---2345----] container curios
3172  k8s-logger:nginx         running  [---2345----] Logger for container nginx
3243  container:nginx          running  [---2345----] container nginx
3176  k8s-logger:web           running  [---2345----] Logger for container web
3219  container:web            running  [---2345----] container web
3181  lldpd                    running  [---2345----] LLDP daemon (IEEE 802.1ab)
3182  staticd                  running  [---2345----] Static routing daemon
admin@dut4:~$ 
```

I also tried no enabled copy ru st, rebooted, when it started all containers was up

### Expected Behavior

Containers should not start when disabled.

### Steps To Reproduce

  "infix-containers:containers": {
    "container": [
      {
        "name": "alpine-nginx",
        "enabled": false,
        "image": "docker.io/yobasystems/alpine-nginx:latest",
        "command": "sleep inf"
      },
      {
        "name": "curios",
        "image": "ghcr.io/kernelkit/curios:24.05.0",
        "command": "/usr/sbin/telnetd  -F",
        "network": {
          "host": true
        }
      },
      {
        "name": "nginx",
        "image": "docker.io/library/nginx",
        "command": "sleep inf"
      },
      {
        "name": "web",
        "image": "oci-archive:curios-httpd-v24.03.0.tar.gz",
        "command": "/usr/sbin/httpd -f -v -p 81",
        "network": {
          "host": true
        }
      }
    ]
  },

troglobit, 2024-06-21:
Reproduced.

troglobit, 2024-06-21:
Root cause is that `container delete`, see below, takes too long and triggers the sysrepo internal timeout:

https://github.com/kernelkit/infix/blob/2b1543ba115c1c24c6043137821942b41a8b0e63/src/confd/src/infix-containers.c#L199-L200

Commenting out that line, keeping `initctl disable`, then everything works perfectly.  I'll look into moving the delete action to our background task manager, `execd`.

commit 563a33c1 (2024-06-24): confd: when disabling containers, delete in background

Instead of potentially blocking sysrepo, which has a 5 second timeout,
we move 'container delete' jobs to be run by the execd queue.

Fixes #479
commit afbe5ca2 (2024-06-25): confd: when disabling containers, delete in background

Instead of potentially blocking sysrepo, which has a 5 second timeout,
we move 'container delete' jobs to be run by the execd queue.

Fixes #479
closed 2024-06-25 by troglobit