kernelkit/infix #503: Configurable syslog support

troglobit, 2024-07-02:
### Requirements

 - Ensure `authpriv.*` is never sent to remote server
 - Configurable log filters, as syslogd supports (_optional_)
 - Log to external media
 - Log to remote server(s)
 - Add default rules to factory-config for local logging to RAM disk

> **Note:** check for available YANG models, and configurable filters really is optional.
> As rev. 1 we could just support logging to external media and remote server(s)

troglobit, 2024-07-10:
Current `ietf-syslog.yang` (draft) model https://datatracker.ietf.org/doc/draft-ietf-netmod-syslog-model/

```
module: ietf-syslog
  +--rw syslog!
     +--rw actions
        +--rw console! {console-action}?
        |  +--rw facility-filter
        |  |  +--rw facility-list* [facility severity]
        |  |     +--rw facility            union
        |  |     +--rw severity            union
        |  |     +--rw advanced-compare {select-adv-compare}?
        |  |        +--rw compare?   enumeration
        |  |        +--rw action?    identityref
        |  +--rw pattern-match?     string {select-match}?
        +--rw file {file-action}?
        |  +--rw log-file* [name]
        |     +--rw name               inet:uri
        |     +--rw facility-filter
        |     |  +--rw facility-list* [facility severity]
        |     |     +--rw facility            union
        |     |     +--rw severity            union
        |     |     +--rw advanced-compare {select-adv-compare}?
        |     |        +--rw compare?   enumeration
        |     |        +--rw action?    identityref
        |     +--rw pattern-match?     string {select-match}?
        |     +--rw structured-data?   boolean {structured-data}?
        |     +--rw file-rotation
        |        +--rw number-of-files?   uint32 {file-limit-size}?
        |        +--rw max-file-size?     uint32 {file-limit-size}?
        |        +--rw rollover?          uint32 {file-limit-duration}?
        |        +--rw retention?         uint32 {file-limit-duration}?
        +--rw remote {remote-action}?
           +--rw destination* [name]
              +--rw name                 string
              +--rw (transport)
              |  +--:(udp)
              |  |  +--rw udp
              |  |     +--rw address?   inet:host
              |  |     +--rw port?      inet:port-number
              |  +--:(tls)
              |     +--rw tls
              |        +--rw address?   inet:host
              |        +--rw port?      inet:port-number
              +--rw facility-filter
              |  +--rw facility-list* [facility severity]
              |     +--rw facility            union
              |     +--rw severity            union
              |     +--rw advanced-compare {select-adv-compare}?
              |        +--rw compare?   enumeration
              |        +--rw action?    identityref
              +--rw pattern-match?       string {select-match}?
              +--rw structured-data?     boolean {structured-data}?
              +--rw facility-override?   identityref
              +--rw source-interface?    if:interface-ref {remote-source-interface}?
              +--rw signing! {signed-messages}?
                 +--rw cert-signers
                    +--rw cert-signer* [name]
                    |  +--rw name              string
                    |  +--rw cert
                    |  +--rw hash-algorithm?   enumeration
                    +--rw cert-initial-repeat?   uint32
                    +--rw cert-resend-delay?     uint32
                    +--rw cert-resend-count?     uint32
                    +--rw sig-max-delay?         uint32
                    +--rw sig-number-resends?    uint32
                    +--rw sig-resend-delay?      uint32
                    +--rw sig-resend-count?      uint32
```

troglobit, 2024-07-10:
Split up work in three parts (feature levels):

- [ ] `{file-action}`
- [ ] `{remote-action}`
- [ ] `{console-action}`

The console action may not even be necessary.

troglobit, 2024-07-11:
ping @jovatn -- maybe you could take a quick peek at this scope proposal?

jovatn, 2024-07-11:
What a nice surprise! :-)

troglobit, 2024-07-13:
Intermediate model before log format augments.  Notice the global file rotation and server mode.

```
$ yanglint -p ../../../output/target/usr/share/yang/ -f tree -F ietf-syslog:file-action,file-limit-size,remote-action ietf-inet-types@2013-07-15.yang ietf-syslog@2024-03-21.yang infix-syslog@2024-07-13.yang 
module: ietf-inet-types

module: ietf-syslog
  +--rw syslog!
     +--rw actions
     |  +--rw file {file-action}?
     |  |  +--rw log-file* [name]
     |  |     +--rw name               inet:uri
     |  |     +--rw facility-filter
     |  |     |  +--rw facility-list* [facility severity]
     |  |     |     +--rw facility    union
     |  |     |     +--rw severity    union
     |  |     +--rw file-rotation
     |  |        +--rw number-of-files?   uint32 {file-limit-size}?
     |  |        +--rw max-file-size?     uint32 {file-limit-size}?
     |  +--rw remote {remote-action}?
     |     +--rw destination* [name]
     |        +--rw name               string
     |        +--rw (transport)
     |        |  +--:(udp)
     |        |     +--rw udp
     |        |        +--rw address?   inet:host
     |        |        +--rw port?      inet:port-number
     |        +--rw facility-filter
     |           +--rw facility-list* [facility severity]
     |              +--rw facility    union
     |              +--rw severity    union
     +--rw infix-syslog:file-rotation
     |  +--rw infix-syslog:number-of-files?   uint32
     |  +--rw infix-syslog:max-file-size?     uint32
     +--rw infix-syslog:server
        +--rw infix-syslog:enabled?   boolean
        +--rw infix-syslog:listen
           +--rw (infix-syslog:listen)?
              +--:(infix-syslog:udp)
                 +--rw infix-syslog:udp* [port]
                    +--rw infix-syslog:port       inet:port-number
                    +--rw infix-syslog:address?   inet:ip-address

module: infix-syslog

  augment /syslog:syslog:
    +--rw file-rotation
    |  +--rw number-of-files?   uint32
    |  +--rw max-file-size?     uint32
    +--rw server
       +--rw enabled?   boolean
       +--rw listen
          +--rw (listen)?
             +--:(udp)
                +--rw udp* [port]
                   +--rw address?   inet:ip-address
                   +--rw port       inet:port-number
```

closed 2024-08-26 by troglobit