kernelkit/infix #510: Operational status support for DNS and NTP

troglobit, 2024-07-02:
See comments to #249, regarding lacking operational status for DNS, NTP as well as regression tests

- [ ] Add operational support for DNS server
- [ ] Add operational support for NTP server
- [ ] Add associated regression tests (1) for manual/regular DNS and NTP and (2) when used by DHCP

As part of the task, please explore/propose/sync what operational data is available and relevant.
For NTP there is likely a lot of info.
For DNS I suppose there is not so much. If DHCP is used to acquire DNS there is at least some dynamic data.

Consider to include domain "search" setting as part of this task. Alternatively, put that in a separate task as it has lower priority.

jovatn, 2024-09-06:
There is some support for DNS status, at least CLI "show dns" command shows current DNS servers assigned via DHCP.
But if a static DNS server is configured, that is not reflected by "show dns", see examples below.

First setting DNS via DHCP.

```
admin@infix-ad-00-00:/> configure
admin@infix-ad-00-00:/config/> set dhcp-client client-if e28
admin@infix-ad-00-00:/config/> leave
admin@infix-ad-00-00:/> show dns
# resolv.conf from e28.conf
nameserver 172.31.11.11 # e28
nameserver 172.31.11.12 # e28

admin@infix-ad-00-00:/> ping www.kth.se
PING www.kth.se (130.237.28.40) 56(84) bytes of data.
64 bytes from ns-vip-01.sys.kth.se (130.237.28.40): icmp_seq=1 ttl=240 time=2.34 ms
64 bytes from ns-vip-01.sys.kth.se (130.237.28.40): icmp_seq=2 ttl=240 time=2.24 ms
64 bytes from ns-vip-01.sys.kth.se (130.237.28.40): icmp_seq=3 ttl=240 time=2.27 ms
^C
--- www.kth.se ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 2.236/2.282/2.338/0.042 ms
admin@infix-ad-00-00:/>
```

Then adding a static DNS server. It used, but not reflected by "show dns".

```
admin@infix-ad-00-00:/> configure 
admin@infix-ad-00-00:/config/> set system dns-resolver server mydns udp-and-tcp address 8.8.8.8
admin@infix-ad-00-00:/config/> leave
admin@infix-ad-00-00:/> show dns
# resolv.conf from e28.conf
nameserver 172.31.11.11 # e28      <== 8.8.8.8 not stated, although used (see tcpdump below)
nameserver 172.31.11.12 # e28

admin@infix-ad-00-00:/> ping www.google.com
PING www.google.com (142.250.74.100) 56(84) bytes of data.
64 bytes from arn11s10-in-f4.1e100.net (142.250.74.100): icmp_seq=1 ttl=115 time=2.42 ms
64 bytes from arn11s10-in-f4.1e100.net (142.250.74.100): icmp_seq=2 ttl=115 time=2.42 ms
^C
--- www.google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 2.419/2.420/2.422/0.001 ms
admin@infix-ad-00-00:/>

user@pc:~$ sudo tcpdump -n -i eth1 udp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
08:43:45.752192 IP 198.18.103.193.54379 > 8.8.8.8.53: 21923+ [1au] A? www.google.com. (43)
08:43:45.752243 IP 198.18.103.193.54379 > 8.8.8.8.53: 27087+ [1au] AAAA? www.google.com. (43)
08:43:45.754379 IP 8.8.8.8.53 > 198.18.103.193.54379: 27087 1/0/1 AAAA 2a00:1450:400f:805::2004 (71)
08:43:45.763555 IP 8.8.8.8.53 > 198.18.103.193.54379: 21923 1/0/1 A 142.250.74.68 (59)
08:43:45.766524 IP 198.18.103.193.50403 > 8.8.8.8.53: 64310+ [1au] PTR? 68.74.250.142.in-addr.arpa. (55)
08:43:45.768912 IP 8.8.8.8.53 > 198.18.103.193.50403: 64310 1/0/1 PTR arn09s23-in-f4.1e100.net. (93)
``

troglobit, 2025-01-24:
Support for showing statically configured DNS servers has been fixed in main since 685b04d7.

troglobit, 2025-01-24:
Proposed YANG extension of ietf-system:system-state

```yang
    container dns-resolver {
      description "List of active DNS servers";
      config false;

      list server {
        key "address";
      
        leaf address {
          description "IP address of DNS server";
          type inet:ip-address;
        }
      
        leaf origin {
          description "How DNS server was acquired";
          type enumeration {
            enum static {
              description "Statically configured";
            }
            enum dhcp {
              description "Dynamically acquired via DHCP";
            }
          }
        }
      
        leaf interface {
          type if:interface-ref;
          description "Interface DNS server was learned from, if DHCP";
        }
      }
  
      container statistics {
        description "DNS resolver statistics";
    
        leaf cache-size {
          description "Current number of entries in DNS cache";
          type yang:counter32;
        }
    
        leaf cache-hits {
          description "Number of successful cache lookups";
          type yang:counter64;
        }
    
        leaf cache-misses {
          description "Number of failed cache lookups";
          type yang:counter64;
        }
      }
    }
```

Tree view:

<pre>
  +--ro system-state
     +--ro platform
     |  +--ro os-name?      string
     |  +--ro os-release?   string
     |  +--ro os-version?   string
     |  +--ro machine?      string
     +--ro clock
     |  +--ro current-datetime?   yang:date-and-time
     |  +--ro boot-datetime?      yang:date-and-time
     +--ro infix-sys:ntp
     |  +--ro infix-sys:sources
     |     +--ro infix-sys:source* [address]
     |        +--ro infix-sys:address    inet:ip-address
     |        +--ro infix-sys:mode?      source-mode
     |        +--ro infix-sys:state?     source-state
     |        +--ro infix-sys:stratum?   uint8
     |        +--ro infix-sys:poll?      uint8
<b>     +--ro infix-sys:dns-resolver
     |  +--ro infix-sys:server* [address]
     |  |  +--ro infix-sys:address      inet:ip-address
     |  |  +--ro infix-sys:origin?      enumeration
     |  |  +--ro infix-sys:interface?   if:interface-ref
     |  +--ro infix-sys:statistics
     |     +--ro infix-sys:cache-size?     yang:counter32
     |     +--ro infix-sys:cache-hits?     yang:counter64
     |     +--ro infix-sys:cache-misses?   yang:counter64</b>
</pre>

commit 28008a0c (2025-01-27): Add support for DNS resolver operational data

Example status output:

    "infix-system:dns-resolver": {
      "options": {
        "timeout": 3,
        "attempts": 5
      },
      "search": [
        "example.com",
        "foo.com"
      ],
      "server": [
        {
          "address": "1.2.3.4",
          "origin": "static"
        },
        {
          "address": "192.168.2.1",
          "origin": "dhcp",
          "interface": "e5"
        }
      ]
    }

Fixes #510
commit 2d6cc31f (2025-01-27): Add support for DNS resolver operational data

Example status output:

    "infix-system:dns-resolver": {
      "options": {
        "timeout": 3,
        "attempts": 5
      },
      "search": [
        "example.com",
        "foo.com"
      ],
      "server": [
        {
          "address": "1.2.3.4",
          "origin": "static"
        },
        {
          "address": "192.168.2.1",
          "origin": "dhcp",
          "interface": "e5"
        }
      ]
    }

Fixes #510
commit 755463b9 (2025-01-27): Add support for DNS resolver operational data

Example status output:

    "infix-system:dns-resolver": {
      "options": {
        "timeout": 3,
        "attempts": 5
      },
      "search": [
        "example.com",
        "foo.com"
      ],
      "server": [
        {
          "address": "1.2.3.4",
          "origin": "static"
        },
        {
          "address": "192.168.2.1",
          "origin": "dhcp",
          "interface": "e5"
        }
      ]
    }

Fixes #510
commit 1f6d84b7 (2025-01-28): Add support for DNS resolver operational data

Example status output:

    "infix-system:dns-resolver": {
      "options": {
        "timeout": 3,
        "attempts": 5
      },
      "search": [
        "example.com",
        "foo.com"
      ],
      "server": [
        {
          "address": "1.2.3.4",
          "origin": "static"
        },
        {
          "address": "192.168.2.1",
          "origin": "dhcp",
          "interface": "e5"
        }
      ]
    }

Fixes #510
closed 2025-01-31 by mattiaswal