kernelkit/infix #542: login: warning missing pam_lastlog.so

troglobit, 2024-07-14:
```
login[2819]: PAM unable to dlopen(/lib/security/pam_lastlog.so): /lib/security/pam_lastlog.so: cannot open shared object file: No such file or directory
```

mattiaswal, 2024-08-02:
lastlog is not enabled in linux-pam in buildroot.
I think lastlog is a good thing, but --enable-lastlog needs to be added to buildroot recipe.

troglobit, 2024-08-03:
Aha, time for some upstream work then!  I'll have a look :-)

mattiaswal, 2024-08-03:
> Aha, time for some upstream work then! I'll have a look :-)

Please also upstream my patch on linux pam to fix path for .pc files when you are already on :fire:

troglobit, 2024-08-04:
Ouch, that one looks a bit nasty, their pkg-config cross-build-wrapper seems to be hard coded to /usr/lib, but iirc there's a reason for certain packages to install to /lib ... could become a messy discussion, but sure I'll look into it at the same time.

troglobit, 2024-08-04:
Upstream patch for `pam_lastlog.so` https://lore.kernel.org/buildroot/20240804095012.878259-1-troglobit@gmail.com/T/#u

mattiaswal, 2024-08-05:
> ./output/host/bin/pkg-config --cflags pam

lazzer@tollan ~/Documents/addiva/infix (main)$ ./output/host/bin/pkg-config --cflags pam
Package pam was not found in the pkg-config search path.
Perhaps you should add the directory containing `pam.pc'
to the PKG_CONFIG_PATH environment variable
Package 'pam', required by 'virtual:world', not found
lazzer@tollan ~/Documents/addiva/infix (main)$

mattiaswal, 2024-08-05:
How do your * find output/ -name.pc * looks like?

My (relevant parts) looks like:
<pre>
output/host/x86_64-buildroot-linux-gnu/sysroot/lib/pkgconfig/pamc.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/lib/pkgconfig/pam.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/lib/pkgconfig/pam_misc.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/dbus-cxx-2.0.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libtirpc.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libnetfilter_cthelper.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libnetfilter_conntrack.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libuev.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/ext2fs.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/avahi-core.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libbsd.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libpcre2-8.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libsyslog.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/glib-2.0.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/popt.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libassuan.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/sysrepo.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/libpcre2-posix.pc
output/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig/json-c.pc
.....
</pre>

troglobit, 2024-08-05:
Interesting, here's my output, notably from Buildroot (recently branched from) master:

```bash
~/src/buildroot(linux-pam-options)$ find output/staging/ -name '*.pc'
output/staging/usr/lib/pkgconfig/libcrypt.pc
output/staging/usr/lib/pkgconfig/libxcrypt.pc
output/staging/usr/lib/pkgconfig/python-3.12-embed.pc
output/staging/usr/lib/pkgconfig/fmt.pc
output/staging/usr/lib/pkgconfig/python3.pc
output/staging/usr/lib/pkgconfig/libsyslog.pc
output/staging/usr/lib/pkgconfig/zlib.pc
output/staging/usr/lib/pkgconfig/pam.pc
output/staging/usr/lib/pkgconfig/pam_misc.pc
output/staging/usr/lib/pkgconfig/python3-embed.pc
output/staging/usr/lib/pkgconfig/spdlog.pc
output/staging/usr/lib/pkgconfig/python-3.12.pc
output/staging/usr/lib/pkgconfig/pamc.pc
output/staging/usr/lib/pkgconfig/expat.pc
output/staging/usr/lib/pkgconfig/openssl.pc
output/staging/usr/lib/pkgconfig/libssl.pc
output/staging/usr/lib/pkgconfig/libffi.pc
output/staging/usr/lib/pkgconfig/libcrypto.pc
```

Lemme have a look around the tree.

troglobit, 2024-08-05:
Found it, root cause is:

```
# BR2_ROOTFS_MERGED_USR is not set
```

With it unset, as in Infix' defconfigs, I get the same as you:

```bash
~/src/buildroot(linux-pam-options)$ find output/staging/ -name '*.pc'
output/staging/usr/lib/pkgconfig/libcrypt.pc
output/staging/usr/lib/pkgconfig/libxcrypt.pc
output/staging/usr/lib/pkgconfig/python-3.12-embed.pc
output/staging/usr/lib/pkgconfig/python3.pc
output/staging/usr/lib/pkgconfig/libsyslog.pc
output/staging/usr/lib/pkgconfig/zlib.pc
output/staging/usr/lib/pkgconfig/python3-embed.pc
output/staging/usr/lib/pkgconfig/python-3.12.pc
output/staging/usr/lib/pkgconfig/expat.pc
output/staging/usr/lib/pkgconfig/openssl.pc
output/staging/usr/lib/pkgconfig/libssl.pc
output/staging/usr/lib/pkgconfig/libffi.pc
output/staging/usr/lib/pkgconfig/libcrypto.pc
output/staging/lib/pkgconfig/pam.pc
output/staging/lib/pkgconfig/pam_misc.pc
output/staging/lib/pkgconfig/pamc.pc
```

Since there's no point really in maintaining a difference beteen `/bin,/sbin` and `/usr/bin,/usr/sbin` et al on Infix, I suggest we update our defconfigs to:

```
BR2_ROOTFS_MERGED_USR=y
```

Which is the default, so dropping the line entirely would be equivalent.

commit 8720bf09 (2024-08-12): configs: enable pam_lastlog.so

Backport pam-lastlog build fix from latest Buildroot master and enable
in all defconfigs to fix syslog warning:

   login[2819]: PAM unable to dlopen(/lib/security/pam_lastlog.so)

Fixes #542
PR #551 (2024-08-16): Update NanoPi R2S
## Description

This PR holds massive updates of the Infix support to the NanoPi R2S, initially added in #271.  From the ChangeLog:

- Update Linux kernel to v6.10.3 and sync defconfig with aarch64
- Workaround `reboot` command "hang" on NanoPi R2S (failure to reboot) by replacing the Rockchip watchdog driver with "softdog"
- Update U-Boot to v2024.07, enable secure boot loading of images
- Rename interfaces to LAN + WAN to match case and LEDs
- Rename images to `infix-r2s$ver.ext`, not same as other aarch64
- Change rootfs to squashfs for enhanced security
- Add RAUC support [... cut]
commit a1ac456d (2024-08-16): configs: enable pam_lastlog.so

Backport pam-lastlog build fix from latest Buildroot master and enable
in all defconfigs to fix syslog warning:

   login[2819]: PAM unable to dlopen(/lib/security/pam_lastlog.so)

Fixes #542
closed 2024-08-16 by wkz