kernelkit/infix #545: Sort loopback first in show interfaces

troglobit, 2024-07-25:
I propose sorting interfaces in `show interfaces` such that `lo` is always displayed first. It is the only interface that all systems will have and also the order in which standard Linux/UNIX tools display interfaces. 

Consider even listing all interfaces in ifindex order.

wkz, 2024-07-29:
Some thoughts:
- `lo` on top sounds good to me
- `ifindex` is roughly equivalent to the hardware probe order, which is rarely the most intuitive order to view stuff in. I don't think we want to rely on that when sorting

Today, we try convey interface hierarchy when possible (bridge ports are listed "under" their bridge, vlans are stacked on top of their lowers, etc). If we want to find some other sort order, we need to think about the implications on this feature. One option would be to have a separate command that lists interfaces with hierarchy indications, and that `show interfaces` outputs a flat table. In that case the sort key could be `iftype` (where we could rank loopbacks highest), with `ifname` as the tiebreaker within each type.

troglobit, 2024-07-30:
Good point, I had forgotten about the weird probe order thing.

I think the current hierarchy view of `show interfaces` is great!  we already have `show ip [brief|detailed|multicast]` command, so maybe that's sufficient for all other use-cases?

commit c00dfa6b (2024-08-14): cli: always sort lo first in 'show interfaces'

This fixes #545
commit c9ef7750 (2024-08-16): cli: always sort lo first in 'show interfaces'

This fixes #545
closed 2024-08-16 by mattiaswal