kernelkit/infix #558: Factory reset may fail

mattiaswal, 2024-08-14:
Failed to do factory reset, seen when running all tests in a suite (with restconf, probably not relevant)
<pre>
Aug 12 19:36:17 dut1 nginx: fe80::a0f2:8aff:fe1e:9f84 - admin [12/Aug/2024:19:36:17 +0000] "POST /restconf/operations/infix-factory-default:factory-default HTTP/1.1" 204 0 "-" "python-requests/2.32.0"
Aug 12 19:36:17 dut1 nginx: 2024/08/12 19:36:17 [info] 3755#0: 36 client fe80::a0f2:8aff:fe1e:9f84 closed keepalive connection
Aug 12 19:36:17 dut1 rousette[3750]: [2024-08-12 19:36:17.766] [rousette] [info] [::1]:53904: GET /restconf/ds/ietf-datastores:operational 
Aug 12 19:36:17 dut1 rousette[3750]: [2024-08-12 19:36:17.779] [rousette] [trace] Authenticated as user admin 
Aug 12 19:36:18 dut1 confd[3174]: auth_cb() path /ietf-system:system/authentication/user/password reqeust_path /
Aug 12 19:36:18 dut1 container[4853]: Container web-docker0 has been removed.
Aug 12 19:36:18 dut1 yanger[4848]: Command '['podman', 'inspect', 'web-docker0']' returned non-zero exit status 125.
Aug 12 19:36:18 dut1 statd[3656]: Traceback (most recent call last):
Aug 12 19:36:18 dut1 statd[3656]:   File "/usr/libexec/statd/yanger", line 8, in <module>
Aug 12 19:36:18 dut1 statd[3656]:     sys.exit(main())
Aug 12 19:36:18 dut1 statd[3656]:              ^^^^^^
Aug 12 19:36:18 dut1 statd[3656]:   File "/usr/lib/python3.11/site-packages/yanger/yanger.py", line 929, in main
Aug 12 19:36:18 dut1 statd[3656]:   File "/usr/lib/python3.11/site-packages/yanger/yanger.py", line 500, in add_container
Aug 12 19:36:18 dut1 statd[3656]:   File "/usr/lib/python3.11/site-packages/yanger/yanger.py", line 474, in container_inspect
Aug 12 19:36:18 dut1 statd[3656]: KeyError: 0
Aug 12 19:36:18 dut1 statd[3656]: Error, running yanger
Aug 12 19:36:18 dut1 statd[3656]: Error adding yanger data
</pre>

troglobit, 2024-08-14:
No clue so far, by auditing the conftainer wrapper script.  The exit code, 125, seems to indicate there's something wrong with podman itself.

https://docs.podman.io/en/latest/markdown/podman-run.1.html#exit-status

troglobit, 2024-08-16:
Turns out this, and other similar problems, are caused by:

 - Too strict `sys.exit(1)` calls in `yanger`, instead of just returning `{}` or `None`
 - Timing issues when creating/starting/restarting containers

I've played with a retry (3 times) of actions and, at most, one call to `restart` fails.  The second try works, seemingly because the system has finished stopping, or creating, the container.

```tap
======================================================= Iteration 148
9PM - Simplicity is the ultimate sophistication 

Starting test 0001-container_basic.py
2024-08-16 13:09:17 # Starting (2024-08-16 13:09:17)
2024-08-16 13:09:17 # host:    host
2024-08-16 13:09:17 #     tgt:     d4a
2024-08-16 13:09:17 #     data:    d4b
2024-08-16 13:09:17 # target:  dut4
2024-08-16 13:09:17 #     mgmt:    e1
2024-08-16 13:09:17 #     data:    e2
2024-08-16 13:09:17 #
2024-08-16 13:09:17 # Probing dut4 on port d4a for IPv6LL mgmt address ...
2024-08-16 13:09:17 # Testing using RESTCONF
2024-08-16 13:09:17 # YANG models downloaded.
2024-08-16 13:09:17 ok 1 - Set up topology and attach to target DUT
2024-08-16 13:09:17 ok 2 - Create web container from bundled OCI image
2024-08-16 13:09:18 ok 3 - Verify web container has started
2024-08-16 13:09:18 ok 4 - Verify web container responds
2024-08-16 13:09:18 # warn - failed restart web (1/3): 400 Client Error: Bad Request for url: https://[fe80::2a0:85ff:fe00:401%d4a]:443/restconf/data/infix-containers:containers/container=web/restart
2024-08-16 13:09:19 ok 5 - Verify web container can be stopped and restarted
2024-08-16 13:09:20 ok 6 - Verify web container still responds
2024-08-16 13:09:20 # Exiting (2024-08-16 13:09:20)
2024-08-16 13:09:20 6..6
```

commit 9a341ef6 (2024-08-22): statd: return empty data from run commands, drop getitem()

These changes are intended to make yanger a bit more robust.  Instead of
bailing out on command errors, we now return empty string or object back
to the callee in run_cmd() and run_json_cmd().

One example where yanger previously exited hard, and dragged with it the
entire session, is when inspecting container status or when attempting
to restart container instances which have yet to be created/stopped.

Fixes #558
commit b436336f (2024-08-22): test: improve robustness of container actions

If a container has not yet stopped/started we may for proto RESTCONF get
"invalid URI" result back for some container actions.  With this change
we allow the action to be retried up to three times before passing on
the error.

In tests on Qemu (x86_64) this happens very rarely and need at most one
retry before succeeding.  Verified by iterating the same basic test over
night (9000+ iterations).

Fixes #558
PR #563 (2024-08-22): Increase robustness of operational datastore
## Description

This PR increases the robustness of the Infix `yanger` helper.  Basically, instead of bailing out with `sys.exit(1)`, causing `statd` to fail and automatically cause a non-working operational datastore, we allow external commands to fail and then return empty objects/arrays to the respective data parsers.

For example, asking the operational datastore for the status of a container before the container has been started properly should not b0rk the datastore, but rather return `None` information.  Same as if listing all containers.

Th [... cut]
commit 8c5defa9 (2024-08-22): statd: return empty data from run commands, drop getitem()

These changes are intended to make yanger a bit more robust.  Instead of
bailing out on command errors, we now return empty string or object back
to the callee in run_cmd() and run_json_cmd().

One example where yanger previously exited hard, and dragged with it the
entire session, is when inspecting container status or when attempting
to restart container instances which have yet to be created/stopped.

Fixes #558
closed 2024-08-22 by troglobit