kernelkit/infix #562: Incorrect target file for gen-version in the failure() function

ahmkar94, 2024-08-15:
### Current Behavior

In the failure() function, the gen-version command is currently writing its output to the wrong directory. The command:
```
gen-version >"$FACTORY_D/20-version.json"
```
is targeting the $FACTORY_D directory. 

### Expected Behavior

It should instead be targeting the $FAILURE_D directory, like the other commands in this function.
```
gen-version >"$FAILURE_D/20-version.json"
```

troglobit, 2024-08-15:
Nice catch, confirmed!

troglobit, 2024-08-22:
We didn't cover this one in today's inbox meeting, but I'm assigning this one to you. Just make sure to include the "Fixes #562" line somewhere in the commit message for this fix :sunglasses:

ahmkar94, 2024-08-23:
> We didn't cover this one in today's inbox meeting, but I'm assigning this one to you, @ahmkar94. Just make sure to include the "Fixes #562" line somewhere in the commit message for this fix 😎

Did that yesterday in the ongoing PR :)

commit 21274aa3 (2024-08-23): confd: fix output directory for gen-version in gen_failure_cfg

Fixes #562
closed 2024-08-23 by mattiaswal