kernelkit/infix #568: CI: Unit tests fail almost every time due to race condition

mattiaswal, 2024-08-19:
There are some race condition between the two build, both claiming container-name infamy0.
<pre>
time="2024-08-19T11:11:27Z" level=warning msg="For using systemd, you may need to log in using a user session"
time="2024-08-19T11:11:27Z" level=warning msg="Alternatively, you can enable lingering with: `loginctl enable-linger 1006` (possibly as root)"
time="2024-08-19T11:11:27Z" level=warning msg="Falling back to --cgroup-manager=cgroupfs"
time="2024-08-19T11:11:27Z" level=warning msg="The cgroupv2 manager is set to systemd but there is no systemd user session available"
time="2024-08-19T11:11:27Z" level=warning msg="For using systemd, you may need to log in using a user session"
time="2024-08-19T11:11:27Z" level=warning msg="Alternatively, you can enable lingering with: `loginctl enable-linger 1006` (possibly as root)"
time="2024-08-19T11:11:27Z" level=warning msg="Falling back to --cgroup-manager=cgroupfs"
time="2024-08-19T11:11:27Z" level=warning msg="The cgroupv2 manager is set to systemd but there is no systemd user session available"
time="2024-08-19T11:11:27Z" level=warning msg="For using systemd, you may need to log in using a user session"
time="2024-08-19T11:11:27Z" level=warning msg="Alternatively, you can enable lingering with: `loginctl enable-linger 1006` (possibly as root)"
time="2024-08-19T11:11:27Z" level=warning msg="Falling back to --cgroup-manager=cgroupfs"
Error: creating container storage: the container name "infamy0" is already in use by 232b7cb04d60de108004bc47016063089f9c[22](https://github.com/kernelkit/infix/actions/runs/10452222167/job/28940216382#step:7:23)71d4a9685e0c5adb948e17248b. You have to remove that container to be able to reuse that name: that name is already in use, or use --replace to instruct Podman to do so.
make[1]: *** [/home/github-lantean/actions-runner/_work/infix/infix/test/test.mk:[24](https://github.com/kernelkit/infix/actions/runs/10452222167/job/28940216382#step:7:25): test-unit] Error 125
make[1]: Leaving directory '/home/github-lantean/actions-runner/_work/infix/infix/buildroot'
make: *** [Makefile:31: test-unit] Error 2
Error: Process completed with exit code 2.
</pre>
See: https://github.com/kernelkit/infix/actions/runs/10452222167/job/28940216382

troglobit, 2024-08-21:
@mattiaswal + @wkz maybe, for the time being, disable unit tests for the aarch64 build?

troglobit, 2024-08-21:
Better fix now on the `test-reset-fail` branch.  Uses random names (default docker behavior) for all CI tests.

commit 180abb65 (2024-08-21): test: run unit tests with a random container name

Fixes #568
commit 6a4ea6d8 (2024-08-22): test: run all CI tests with a random container name

Fixes #568
commit 063c49c3 (2024-08-22): test: run all CI tests with a random container name

Fixes #568
commit 588d8189 (2024-08-22): test: run all CI tests with a random container name

Fixes #568
closed 2024-08-22 by troglobit