kernelkit/infix #580: Handle breaking changes (backup/restore)

troglobit, 2024-08-22:
Infix recently gained support for handling breaking model changes, with a JSON transformation framework for `startup-config` at boot.

However, many users want to be able to keep backups of device configurations to restore when, e.g., replacing a faulty unit, replicating a setup elsewhere, or during troubleshooting. These backups may not be compatible with the device's firmware version.

To improve the UX of this possible mess, dedicated RPCs for backup/restore are proposed.  The backup RPC is more for symmetry, the real action is behind the restore of a possibly incompatible old backup configuration, where the same JSON transformations need to be done, and handled properly on failure.

troglobit, 2024-08-22:
Another use-case: user can verify an older .cfg file can be "played forward" and migrated to the latest syntax.

troglobit, 2024-10-01:
Feature future.

jovatn, 2026-08-17:
CCB: Good to include, although no rush. Future.