kernelkit/infix #584: Etherlike interfaces that appear after initial configuration is created is handled as ethernet interfaces

mattiaswal, 2024-08-22:
In qemu:
Have a startup configuration with only one interface.
Change number of interfaces, the new intefaces will now be handled as ethernet interfaces. With this as result when trying to configure the new interface:
<pre>
Aug 22 13:24:14 infix-00-00-00 klishd: Service process for client was forked: 3549
Aug 22 13:24:14 infix-00-00-00 klishd[3479]: klishd: Service process for client was forked: 3549
Aug 22 13:24:14 infix-00-00-00 klishd-service: Start SysRepo session for "admin"
Aug 22 13:24:14 infix-00-00-00 klishd[3479]: klishd-service: Start SysRepo session for "admin"
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Interface e1.IPv6 no longer relevant for mDNS.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Leaving mDNS multicast group on interface e1.IPv6 with address fe80::ff:fe00:1.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Interface e1.IPv4 no longer relevant for mDNS.
Aug 22 13:27:13 infix-00-00-00 kernel: 8021q: adding VLAN 0 to HW filter on device e1
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Leaving mDNS multicast group on interface e1.IPv4 with address 192.168.50.1.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Withdrawing address record for fe80::ff:fe00:1 on e1.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Withdrawing address record for 192.168.50.1 on e1.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Joining mDNS multicast group on interface e1.IPv4 with address 192.168.50.1.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: New relevant interface e1.IPv4 for mDNS.
Aug 22 13:27:13 infix-00-00-00 avahi-daemon[3493]: Registering new address record for 192.168.50.1 on e1.IPv4.
<b>
Aug 22 13:27:13 infix-00-00-00 dagger[3069]: Aborting: /run/net/1/action/init/e2/10-ethtool-aneg.sh failed with exitcode 75
Aug 22 13:27:13 infix-00-00-00 dagger[3069]: Abandoned generation 1
Aug 22 13:27:13 infix-00-00-00 confd[3069]: Failed to apply interface configuration
Aug 22 13:27:13 infix-00-00-00 confd[3069]: [ERR] Oups, error detected in SR_EV_DONE 
</b>
Aug 22 13:27:22 infix-00-00-00 klishd-service: Stop SysRepo session for "admin"
Aug 22 13:27:22 infix-00-00-00 klishd[3479]: klishd-service: Stop SysRepo session for "admin"
</pre>

troglobit, 2024-08-22:
Would be great to support this a bit better. Currently, when I do the same, I just do a factory reset.

mattiaswal, 2024-08-22:
> Would be great to support this a bit better. Currently, when I do the same, I just do a factory reset.

Yes, no blocker bug, but irritating.

wkz, 2024-10-01:
### Steps to reproduce

(Presumably)

1. `./images/qemu.sh -c`
1. Choose TAP networking, 1 port
1. `./images/qemu.sh`
1. Shutdown
1. `./images/qemu.sh -c`
1. Choose TAP networking, 2 ports
1. `./images/qemu.sh`
1. `configure` `set interface e2` `leave`

Since `e2` is not in running, the old `confd` logic (which did not discriminate between ethernet and etherlike) will kick in, configuring the interface as `ethernet`. This then triggers `ethtool` setup, which fails.

### Proposal

Use the same iftype resolution logic in `confd` as in `gen-interfaces`.

commit 470055ed (2025-02-13): Remove interface type Etherlike

Instead check if the interface support autonegotion
before setting it to an interface.

This fix #877 and fix #584
commit b290e443 (2025-02-16): Remove interface type Etherlike

Instead check if the interface support autonegotion
before setting it to an interface.

This fix #877 and fix #584
closed 2025-02-14 by wkz