kernelkit/infix #585: Failure in bridge configuration lock sysrepo

mattiaswal, 2024-08-22:
Very easy to reproduce. I use qemu with 6 interfaces

<pre>
admin@infix-00-00-00:/> copy fa ru
admin@infix-00-00-00:/> configure 
admin@infix-00-00-00:/config/> set interface br1
admin@infix-00-00-00:/config/> set interface e3 bridge-port bridge br1
admin@infix-00-00-00:/config/> set interface br1 bridge vlans vlan 30 multicast-filter 224.2.2.2 ports e3
admin@infix-00-00-00:/config/> leave 
Error: EV ORIGIN: SHM event "done" ID 2 processing timed out.
Please check your changes, try 'diff' and 'do show interfaces'.
admin@infix-00-00-00:/config/> 
admin@infix-00-00-00:/config/> abort
admin@infix-00-00-00:/> reboot
Error: EV ORIGIN: SHM event "rpc" ID 1 processing timed out.
Error: failed sending RPC /ietf-system:system-restart: User callback failed
admin@infix-00-00-00:/> 
</pre>

This config is wrong (e3 does not exist in vlan 30) and should be blocked by a must expression, but still this feel fishy.

mattiaswal, 2024-08-22:
Failed even with correct VLAN setting (e3 untagged).
If i do it in small steps it succeed. Configure one thing and leave.

troglobit, 2024-08-23:
Reproduced. Have possible sysrepo fix in my tree.

commit df452fc2 (2024-08-27): patches/sysrepo: exit sysrepo-plugind on thread processing error

When sysrepo-plugind is processing callback events, make sure to log the
error and exit(1) the entire process so the system can handle the problem.
For example, restart all dependent services.

Fixes #585
commit e1c48102 (2024-08-28): patches/sysrepo: exit sysrepo-plugind on thread processing error

When sysrepo-plugind is processing callback events, make sure to log the
error and exit(1) the entire process so the system can handle the problem.
For example, restart all dependent services.

Fixes #585
commit dfcd8a7a (2024-08-28): patches/sysrepo: exit sysrepo-plugind on thread processing error

When sysrepo-plugind is processing callback events, make sure to log the
error and exit(1) the entire process so the system can handle the problem.
For example, restart all dependent services.

Fixes #585
closed 2024-08-28 by troglobit