kernelkit/infix #636: OSPF database support (ietf-ospf.yang)

jovatn, 2024-09-16:
### Description

List OSPF database via NETCONF is not supported, as the associated yang branch of "ietf-ospf.yang" is deviated as not supported by "infix-routing.yang".
This feature request concerns listing what is currently available via CLI "show ospf database" also via YANG.

### Additional Information

The database branch of the "ietf-ospf" yang model is huge. Determining what is supported by Infix today is the first step in order to suggest the scope for this feature request.
An rudimentary example of what can be shown by CLI today is given below.

```
admin@R6:/> show ospf database 

       OSPF Router with ID (10.1.1.6)

                Router Link States (Area 0.0.0.0)

Link ID         ADV Router      Age  Seq#       CkSum  Link count
10.1.1.1       10.1.1.1        1523 0x80000007 0x03ce 3
10.1.1.2       10.1.1.2        1494 0x80000009 0x366c 4
10.1.1.3       10.1.1.3        1493 0x80000009 0xaeed 4
10.1.1.4       10.1.1.4        1502 0x80000007 0x596d 3
10.1.1.5       10.1.1.5        1561 0x80000009 0xb4e4 4
10.1.1.6       10.1.1.6        1510 0x80000009 0xe4ae 4

admin@R6:/> 
```

### General Information

Anyone can help out by [sponsoring][1] development of new features or [contributing][2] pull requests.
Please use this issue for discussions related to the feature.

[1]: https://github.com/kernelkit/infix/blob/main/.github/SUPPORT.md
[2]: https://github.com/kernelkit/infix/blob/main/.github/CONTRIBUTING.md

jovatn, 2024-09-16:
I include a discussion proposal for infix-routing.yang for ospf database support in infix.
[infix-routing@2024-09-16.yang.txt](https://github.com/user-attachments/files/17014159/infix-routing%402024-09-16.yang.txt)

A diff towards the current infix-routing yang is also provided
[diff-ospf-database.txt](https://github.com/user-attachments/files/17014248/diff-ospf-database.txt)

In short, the idea is to open up support for OSPF database in "ietf-ospf.yang" for OSPFv2, but skipping any "opaque LSA" support. Hopefully this matches what FRR gives us. See the "ro ospf:database" branch in the pyang output below.

```
PC :~infix/src/confd/yang (main)$ pyang -f tree ietf-ospf@2022-10-19.yang ietf-routing@2018-03-13.yang infix-routing@2024-09-16.yang
infix-routing@2024-09-16.yang:20: warning: imported module "ietf-interfaces" not used
infix-routing@2024-09-16.yang:24: warning: imported module "ietf-routing-types" not used
module: ietf-routing
  +--rw routing
     +--rw control-plane-protocols
     |  +--rw control-plane-protocol* [type name]
     |     +--rw type             identityref
     |     +--rw name             string
     |     +--rw static-routes
     |     |  +--rw v4ur:ipv4
     |     |  |  +--rw v4ur:route* [destination-prefix]
     |     |  |     +--rw v4ur:destination-prefix    inet:ipv4-prefix
     |     |  |     +--rw v4ur:description?          string
     |     |  |     +--rw v4ur:next-hop
     |     |  |        +--rw (v4ur:next-hop-options)
     |     |  |           +--:(v4ur:simple-next-hop)
     |     |  |           |  +--rw v4ur:outgoing-interface?   if:interface-ref
     |     |  |           |  +--rw v4ur:next-hop-address?     inet:ipv4-address
     |     |  |           +--:(v4ur:special-next-hop)
     |     |  |              +--rw v4ur:special-next-hop?     enumeration
     |     |  +--rw v6ur:ipv6
     |     |     +--rw v6ur:route* [destination-prefix]
     |     |        +--rw v6ur:destination-prefix    inet:ipv6-prefix
     |     |        +--rw v6ur:description?          string
     |     |        +--rw v6ur:next-hop
     |     |           +--rw (v6ur:next-hop-options)
     |     |              +--:(v6ur:simple-next-hop)
     |     |              |  +--rw v6ur:outgoing-interface?   if:interface-ref
     |     |              |  +--rw v6ur:next-hop-address?     inet:ipv6-address
     |     |              +--:(v6ur:special-next-hop)
     |     |                 +--rw v6ur:special-next-hop?     enumeration
     |     +--rw ospf:ospf
     |        +--rw ospf:address-family?               iana-rt-types:address-family
     |        +--rw ospf:explicit-router-id?           rt-types:router-id {explicit-router-id}?
     |        +--ro ospf:router-id?                    rt-types:router-id
     |        +--ro ospf:local-rib
     |        |  +--ro ospf:route* [prefix]
     |        |     +--ro ospf:prefix        inet:ip-prefix
     |        |     +--ro ospf:next-hops
     |        |     |  +--ro ospf:next-hop* []
     |        |     |     +--ro ospf:outgoing-interface?   if:interface-ref
     |        |     |     +--ro ospf:next-hop?             inet:ip-address
     |        |     +--ro ospf:route-type?   route-type
     |        +--ro ospf:database
     |        |  +--ro ospf:as-scope-lsa-type* [lsa-type]
     |        |     +--ro ospf:lsa-type         uint16
     |        |     +--ro ospf:as-scope-lsas
     |        |        +--ro ospf:as-scope-lsa* [lsa-id adv-router]
     |        |           +--ro ospf:lsa-id              union
     |        |           +--ro ospf:adv-router          rt-types:router-id
     |        |           +--ro ospf:decode-completed?   boolean
     |        |           +--ro ospf:raw-data?           yang:hex-string
     |        |           +--ro (ospf:version)?
     |        |              +--:(ospf:ospfv2)
     |        |                 +--ro ospf:ospfv2
     |        |                    +--ro ospf:header
     |        |                    |  +--ro ospf:lsa-options
     |        |                    |  |  +--ro ospf:lsa-options*   identityref
     |        |                    |  +--ro ospf:lsa-id         yang:dotted-quad
     |        |                    |  +--ro ospf:age            uint16
[... 81 lines cut ...]
           +--rw name              string
           +--rw address-family    identityref
           +--ro routes
              +--ro route* []
                 +--ro route-preference?          route-preference
                 +--ro next-hop
                 |  +--ro (next-hop-options)
                 |     +--:(simple-next-hop)
                 |     |  +--ro outgoing-interface?      if:interface-ref
                 |     |  +--ro v4ur:next-hop-address?   inet:ipv4-address
                 |     |  +--ro v6ur:next-hop-address?   inet:ipv6-address
                 |     +--:(special-next-hop)
                 |     |  +--ro special-next-hop?        enumeration
                 |     +--:(next-hop-list)
                 |        +--ro next-hop-list
                 |           +--ro next-hop* []
                 |              +--ro outgoing-interface?   if:interface-ref
                 |              +--ro v4ur:address?         inet:ipv4-address
                 |              +--ro v6ur:address?         inet:ipv6-address
                 +--ro source-protocol            identityref
                 +--ro v4ur:destination-prefix?   inet:ipv4-prefix
                 +--ro v6ur:destination-prefix?   inet:ipv6-prefix

PC :~/infix/src/confd/yang (main)$
```

troglobit, 2024-09-22:
Also relevant, issue #647