kernelkit/infix #659: CLI: Show interfaces does not always show all interfaces

mattiaswal, 2024-09-27:
`admin@infix-ad-02-20:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv6                   ::1/128 (static)
dsa0            ethernet   UP          02:00:de:ad:02:20                        
e1              ethernet   UP          02:00:de:ad:02:21                        
                ipv6                   fe80::deff:fead:221/64 (link-layer)
e2              ethernet   UP          02:00:de:ad:02:22                        
                ipv6                   fe80::deff:fead:222/64 (link-layer)
e3              ethernet   UP          02:00:de:ad:02:23                        
                ipv6                   fe80::deff:fead:223/64 (link-layer)
e4              ethernet   UP          02:00:de:ad:02:24                        
                ipv6                   fe80::deff:fead:224/64 (link-layer)
e5              ethernet   UP          02:00:de:ad:02:25                        
                ipv6                   fe80::deff:fead:225/64 (link-layer)
e6              ethernet   UP          02:00:de:ad:02:26                        
                ipv6                   fe80::deff:fead:226/64 (link-layer)
e7              ethernet   UP          02:00:de:ad:02:27                        
                ipv6                   fe80::deff:fead:227/64 (link-layer)
e8              ethernet   UP          02:00:de:ad:02:28                        
                ipv6                   fe80::deff:fead:228/64 (link-layer)
e9              ethernet   UP          02:00:de:ad:02:29                        
                ipv6                   fe80::deff:fead:229/64 (link-layer)
e10             ethernet   UP          02:00:de:ad:02:2a                        
                ipv6                   fe80::deff:fead:22a/64 (link-layer)
e11             ethernet   UP          02:00:de:ad:02:2b                        
                ipv6                   fe80::deff:fead:22b/64 (link-layer)
e12             ethernet   UP          02:00:de:ad:02:2c                        
                ipv6                   fe80::deff:fead:22c/64 (link-layer)
e13             ethernet   UP          02:00:de:ad:02:2d                        
                ipv6                   fe80::deff:fead:22d/64 (link-layer)
e14             ethernet   UP          02:00:de:ad:02:2e                        
                ipv6                   fe80::deff:fead:22e/64 (link-layer)
e15             ethernet   LOWER-DOWN  02:00:de:ad:02:2f                        
e16             ethernet   LOWER-DOWN  02:00:de:ad:02:30                        
e17             ethernet   UP          02:00:de:ad:02:31                        
                ipv6                   fe80::deff:fead:231/64 (link-layer)
e18             ethernet   UP          02:00:de:ad:02:32                        
                ipv6                   fe80::deff:fead:232/64 (link-layer)
e19             ethernet   UP          02:00:de:ad:02:33                        
                ipv6                   fe80::deff:fead:233/64 (link-layer)
e20             ethernet   UP          02:00:de:ad:02:34                        
                ipv6                   fe80::deff:fead:234/64 (link-layer)
e21             ethernet   UP          02:00:de:ad:02:35                        
                ipv6                   fe80::deff:fead:235/64 (link-layer)
e22             ethernet   UP          02:00:de:ad:02:36                        
                ipv6                   fe80::deff:fead:236/64 (link-layer)
e23             ethernet   UP          02:00:de:ad:02:37                        
                ipv6                   fe80::deff:fead:237/64 (link-layer)
admin@infix-ad-02-20:/> 
`
but
`
admin@infix-ad-02-20:~$ sysrepocfg -f json -X -d operational -m ietf-interfaces |/usr/libexec/statd/cli-pretty "show-interfaces"
INTERFACE       PROTOCOL   STATE       DATA                                     
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv6                   ::1/128 (static)
dsa0            ethernet   UP          02:00:de:ad:02:20                        
e1              ethernet   UP          02:00:de:ad:02:21                        
                ipv6                   fe80::deff:fead:221/64 (link-layer)
e2              ethernet   UP          02:00:de:ad:02:22                        
                ipv6                   fe80::deff:fead:222/64 (link-layer)
e3              ethernet   UP          02:00:de:ad:02:23                        
                ipv6                   fe80::deff:fead:223/64 (link-layer)
e4              ethernet   UP          02:00:de:ad:02:24                        
                ipv6                   fe80::deff:fead:224/64 (link-layer)
e5              ethernet   UP          02:00:de:ad:02:25                        
                ipv6                   fe80::deff:fead:225/64 (link-layer)
e6              ethernet   UP          02:00:de:ad:02:26                        
                ipv6                   fe80::deff:fead:226/64 (link-layer)
e7              ethernet   UP          02:00:de:ad:02:27                        
                ipv6                   fe80::deff:fead:227/64 (link-layer)
e8              ethernet   UP          02:00:de:ad:02:28                        
                ipv6                   fe80::deff:fead:228/64 (link-layer)
e9              ethernet   UP          02:00:de:ad:02:29                        
                ipv6                   fe80::deff:fead:229/64 (link-layer)
e10             ethernet   UP          02:00:de:ad:02:2a                        
                ipv6                   fe80::deff:fead:22a/64 (link-layer)
e11             ethernet   UP          02:00:de:ad:02:2b                        
                ipv6                   fe80::deff:fead:22b/64 (link-layer)
e12             ethernet   UP          02:00:de:ad:02:2c                        
                ipv6                   fe80::deff:fead:22c/64 (link-layer)
e13             ethernet   UP          02:00:de:ad:02:2d                        
                ipv6                   fe80::deff:fead:22d/64 (link-layer)
e14             ethernet   UP          02:00:de:ad:02:2e                        
                ipv6                   fe80::deff:fead:22e/64 (link-layer)
e15             ethernet   LOWER-DOWN  02:00:de:ad:02:2f                        
e16             ethernet   LOWER-DOWN  02:00:de:ad:02:30                        
e17             ethernet   UP          02:00:de:ad:02:31                        
                ipv6                   fe80::deff:fead:231/64 (link-layer)
e18             ethernet   UP          02:00:de:ad:02:32                        
                ipv6                   fe80::deff:fead:232/64 (link-layer)
e19             ethernet   UP          02:00:de:ad:02:33                        
                ipv6                   fe80::deff:fead:233/64 (link-layer)
e20             ethernet   UP          02:00:de:ad:02:34                        
                ipv6                   fe80::deff:fead:234/64 (link-layer)
e21             ethernet   UP          02:00:de:ad:02:35                        
                ipv6                   fe80::deff:fead:235/64 (link-layer)
e22             ethernet   UP          02:00:de:ad:02:36                        
                ipv6                   fe80::deff:fead:236/64 (link-layer)
e23             ethernet   UP          02:00:de:ad:02:37                        
                ipv6                   fe80::deff:fead:237/64 (link-layer)
e24             ethernet   UP          02:00:de:ad:02:38                        
                ipv6                   fe80::deff:fead:238/64 (link-layer)
e25             ethernet   UP          02:00:de:ad:02:39                        
                ipv6                   fe80::deff:fead:239/64 (link-layer)
e26             ethernet   UP          02:00:de:ad:02:3a                        
                ipv6                   fe80::deff:fead:23a/64 (link-layer)
e27             ethernet   LOWER-DOWN  02:00:de:ad:02:3b                        
e28             ethernet   UP          02:00:de:ad:02:3c                        
                ipv6                   fe80::deff:fead:23c/64 (link-layer)
eth1            ethernet   DOWN        02:00:de:ad:02:20             
`

jovatn, 2024-09-27:
Has to do something with the length of the output. If I disable ipv6 on interface e1..e6, we can see some more interfaces.
(Disregard that I unplugged  e1..e4, that only affects status, not number of lines)

```
admin@infix-ad-02-20:/> show interfaces 
INTERFACE       PROTOCOL   STATE       DATA                                     
lo              ethernet   UP          00:00:00:00:00:00                        
                ipv4                   127.0.0.1/8 (static)
                ipv6                   ::1/128 (static)
dsa0            ethernet   UP          02:00:de:ad:02:20                        
e1              ethernet   DOWN        02:00:de:ad:02:21                        
e2              ethernet   DOWN        02:00:de:ad:02:22                        
e3              ethernet   DOWN        02:00:de:ad:02:23                        
e4              ethernet   DOWN        02:00:de:ad:02:24                        
e5              ethernet   UP          02:00:de:ad:02:25                        
e6              ethernet   UP          02:00:de:ad:02:26                        
e7              ethernet   UP          02:00:de:ad:02:27                        
                ipv6                   fe80::deff:fead:227/64 (link-layer)
e8              ethernet   UP          02:00:de:ad:02:28                        
                ipv6                   fe80::deff:fead:228/64 (link-layer)
e9              ethernet   UP          02:00:de:ad:02:29                        
                ipv6                   fe80::deff:fead:229/64 (link-layer)
e10             ethernet   UP          02:00:de:ad:02:2a                        
                ipv6                   fe80::deff:fead:22a/64 (link-layer)
e11             ethernet   UP          02:00:de:ad:02:2b                        
                ipv6                   fe80::deff:fead:22b/64 (link-layer)
e12             ethernet   UP          02:00:de:ad:02:2c                        
                ipv6                   fe80::deff:fead:22c/64 (link-layer)
e13             ethernet   UP          02:00:de:ad:02:2d                        
                ipv6                   fe80::deff:fead:22d/64 (link-layer)
e14             ethernet   UP          02:00:de:ad:02:2e                        
                ipv6                   fe80::deff:fead:22e/64 (link-layer)
e15             ethernet   LOWER-DOWN  02:00:de:ad:02:2f                        
e16             ethernet   LOWER-DOWN  02:00:de:ad:02:30                        
e17             ethernet   UP          02:00:de:ad:02:31                        
                ipv6                   fe80::deff:fead:231/64 (link-layer)
e18             ethernet   UP          02:00:de:ad:02:32                        
                ipv6                   fe80::deff:fead:232/64 (link-layer)
e19             ethernet   UP          02:00:de:ad:02:33                        
                ipv6                   fe80::deff:fead:233/64 (link-layer)
e20             ethernet   UP          02:00:de:ad:02:34                        
                ipv6                   fe80::deff:fead:234/64 (link-layer)
e21             ethernet   UP          02:00:de:ad:02:35                        
                ipv6                   fe80::deff:fead:235/64 (link-layer)
e22             ethernet   UP          02:00:de:ad:02:36                        
                ipv6                   fe80::deff:fead:236/64 (link-layer)
e23             ethernet   UP          02:00:de:ad:02:37                        
                ipv6                   fe80::deff:fead:237/64 (link-layer)
e24             ethernet   UP          02:00:de:ad:02:38                        
                ipv6                   fe80::deff:fead:238/64 (link-layer)
e25             ethernet   UP          02:00:de:ad:02:39                        
                ipv6                   fe80::deff:fead:239/64 (link-layer)
e26             ethernet   UP          02:00:de:ad:02:3a                        
admin@infix-ad-02-20:/> 
```

jovatn, 2024-09-27:
The issue only appears when accessing CLI via console.
When accessing CLI via ssh, all interfaces are shown.

troglobit, 2024-10-31:
CCB decision: klish misbehaves with resize and on console which in turn messes up the pager.   Idea, disabling paging by default in klish.xml and use explicit paging for commands that need it "show log foo".

troglobit, 2024-11-17:
Debugging shows there's a combination of problems, the pager (`less`) wants total control of the terminal, but the mode we run the command `sysrepocfg -X -f json | jq -C .` was in klish "native" terminal mode.  Compare this to the `text-editor` command, which works perfectly because it runs in klish "raw" terminal mode.   Simply updating /etc/klish/infix.xml to set `out="tty"` goes a long way, but with a few backported patches to klish (adjustments to raw mode), we get all the way.  In raw mode the commands need to pipe to a pager of their choice manually.

PR incoming.

commit 46dd0c74 (2024-11-18): src/klish-plugin-infix: use shell pager instead of klish pager

With interactive mode (out="tty") klish puts the tty in raw mode which
fixes all weird issues reported around paged output in the CLI.

Fixes #659
closed 2024-11-18 by troglobit