kernelkit/infix #660: test: uninitalized node attributes warning

troglobit, 2024-09-27:
Very annoying bug/omission in physical topology.

```
$ make test-sh
Info: Generating topology
Info: Generating node YAML
gvpr: warning: Using value of uninitialized node attribute "qn_basemac" of "dut1"
gvpr: warning: Using value of uninitialized edge attribute "qn_headmac" of "dut1--dut2"
gvpr: warning: Using value of uninitialized edge attribute "qn_tailmac" of "dut1--dut2"
Info: Generating executables
Info: Launching dut1
Info: Launching dut2
Info: Launching dut3
Info: Launching dut4
```

ahmkar94, 2024-09-27:
The warnings have been resolved in the `qeneth` project. Integration/Review is pending @wkz , allowing the correct revision to be referenced.

troglobit, 2024-09-27:
Nice work!  (For future reference, it's really helpful if you drop a link to projects outside of the issue you report in :)

troglobit, 2024-09-28:
I see @wkz merged this tonight.  Please post a submodule update as well so we can close this issue properly :-)

commit 2343cfcf (2024-09-30): test/qeneth: Update submodule

Fixes #660
commit 1531688a (2024-09-30): test/qeneth: Update submodule

Fixes #660
closed 2024-09-30 by ahmkar94