kernelkit/infix #678: mdns: add support for deny-interfaces

troglobit, 2024-10-01:
Title says it all.

troglobit, 2024-11-20:
Stretch goal for v24.11.1

troglobit, 2024-11-24:
### Discuss

Plural vs singular form of `allow-interfaces`, `deny-interfaces`, `reflect-filters`, vs s: `allow-interface`, `deny-interface`, `reflect-filter`.  I've used different styles in the proposal above to see if anyone picks up on it.

Consider the following CLI input:

```
admin@example:/config/mdns/> set allow-interfaces e1
admin@example:/config/mdns/> set allow-interfaces e2
admin@example:/config/mdns/> set allow-interfaces e3
```

vs the more concise:

```
admin@example:/config/mdns/> set allow-interface e1
admin@example:/config/mdns/> set allow-interface e2
admin@example:/config/mdns/> set allow-interface e3
```

jovatn, 2024-11-25:
I spot "defaullt" in 'leaf domain' and 'leaf-list reflect-filter'.
@troglobit, is there a default for 'leaf reflector'? true or false?

jovatn, 2024-11-25:
I have no strong opinion on the plural or singular question. Do we have some similar setting to compare with? Perhaps "set client-if eth0" for DHCP client?

troglobit, 2024-11-25:
> I spot "defaullt" in 'leaf domain' and 'leaf-list reflect-filter', is there a default for 'leaf reflector'? true or false?

The default is `false`, similar to the `enabled` leaf.  But I get what you're going for, in the case of our products maybe `true` would be a better default?

troglobit, 2024-11-25:
> I have no strong opinion on the plural or singular question. Do we have some similar setting to compare with? Perhaps "set client-if eth0" for DHCP client?

True, also a homegrown model, although loosely based on the IETF DHCPv6 model(s).  Maybe the DHCP `option` list is also worth mentioning here as an example.

commit 72c2d494 (2024-11-24): confd: expose more avahi-daemon settings in mdns service context

Fix #678
commit 529b422a (2024-12-03): confd: expose more avahi-daemon settings in mdns service context

Use container for interface allow/deny and for reflector settings.
Hopefully there will be more settings in the future, e.g., control
reflector interfaces independently.

Fix #678
commit f0aac343 (2024-12-04): confd: expose more avahi-daemon settings in mdns service context

Use container for interface allow/deny and for reflector settings.
Hopefully there will be more settings in the future, e.g., control
reflector interfaces independently.

Fix #678
commit 8c49c114 (2024-12-04): confd: expose more avahi-daemon settings in mdns service context

Use container for interface allow/deny and for reflector settings.
Hopefully there will be more settings in the future, e.g., control
reflector interfaces independently.

Fix #678
closed 2024-12-04 by wkz