kernelkit/infix #687: Refactor upgrade test for generic file server

troglobit, 2024-10-07:
The upgrade test currently has a FileServer class that could be refactored into a generic class for other tests.  This would allow Infix DUTs to download whatever they want from "the Internet", e.g., a too big container image, or customer specific test data.

For this to be properly versioned/integrated and reproducible (think: `make sdk`), the big container image, or customer specific test data, would be handled by a generic `package/foo/foo.mk` (not host package!) that employs Buildroot's download and hash verification before installing the data, *not to output/target/`, but to `$BINARIES_DIR`, which the test could then map into the Infamy docker and serve as a file area.

troglobit, 2024-10-31:
CCB decision: wkz says, remember to restore boot order after test.  (Test system relies on netbooting)

mattiaswal, 2024-12-06:
This is done in #846

closed 2024-12-06 by mattiaswal