kernelkit/infix #698: Odd log messages: failed giving admin UNIX wheel permissions

troglobit, 2024-10-10:
From `/log/syslog` on a recent main (c44621b8), the following odd log messages can be found:

<pre>
Oct 10 17:07:48 ix finit[1]: Starting confd[3047]
Oct 10 17:07:48 ix load[3055]: Loading startup-config ...
Oct 10 17:07:48 ix confd[3047]: Username "admin" is valid
<b>Oct 10 17:07:48 ix confd[3047]: Failed giving user "admin" UNIX sys-cli permissions.
Oct 10 17:07:48 ix confd[3047]: Failed giving user "admin" UNIX wheel permissions.</b>
Oct 10 17:07:48 ix confd[3047]: Reusing uid:gid 1000:1000 and /home/admin for new user "admin"
<b>Oct 10 17:07:48 ix confd[3047]: User "admin" added to UNIX "sys-cli" group.
Oct 10 17:07:48 ix confd[3047]: User "admin" created, mapped to existing home directory.</b>
Oct 10 17:07:48 ix confd[3047]: Password updated for user "admin"
Oct 10 17:07:48 ix confd[3047]: Login shell updated for user "admin"
Oct 10 17:07:48 ix confd[3047]: addgroup: unknown user admin
Oct 10 17:07:48 ix last message buffered 1 times
Oct 10 17:07:48 ix load[3055]: Success, syncing with startup datastore.
Oct 10 17:07:48 ix load[3055]: Loaded /cfg/startup-config.cfg successfully.
</pre>

> The first **bold** messages in particular.

mattiaswal, 2024-10-11:
Failed to reproduce this over night, propose closing and if it still there, please add steps to reproduce (startup configuration is a good start)

troglobit, 2024-10-15:
Agreed, I too have not seen this anymore.

closed 2024-10-15 by troglobit