kernelkit/infix #730: NTP configuration not shown in CLI

axkar, 2024-10-17:
Navigate to `cli`, enable `ntp`, and `show` the `ntp` configuration

```
admin@infix-00-00-00:/config/> set system ntp enabled 
admin@infix-00-00-00:/config/> set system ntp server pool udp address pool.ntp.org
admin@infix-00-00-00:/config/> diff
system {
+  ntp {
+    enabled true;
+    server pool {
+      udp {
+        address pool.ntp.org;
+      }
+    }
+  }
}
admin@infix-00-00-00:/config/> leave 
admin@infix-00-00-00:/> show ntp 
NTP client not enabled.
```

Obs! Ensure that dhcp client is enabled:
It is verified that the `chronyd` is started properly.

axkar, 2024-10-17:
Additional findings:
By running little check for `chronyc serverstats` as `admin` user, the following error shows up:
```
admin@infix-00-00-00:~$ chronyc serverstats
501 Not authorised
```
And running the same command as `root` works fine:
```
admin@infix-00-00-00:~$ sudo chronyc serverstats
NTP packets received       : 0
NTP packets dropped        : 0
Command packets received   : 11
Command packets dropped    : 0
Client log records dropped : 0
NTS-KE connections accepted: 0
NTS-KE connections dropped : 0
Authenticated NTP packets  : 0
Interleaved NTP packets    : 0
NTP timestamps held        : 0
NTP timestamp span         : 0
NTP daemon RX timestamps   : 0
NTP daemon TX timestamps   : 0
NTP kernel RX timestamps   : 0
NTP kernel TX timestamps   : 0
NTP hardware RX timestamps : 0
NTP hardware TX timestamps : 0
```

Similar for `ntp`  as admin:
```
admin@infix-00-00-00:/> show ntp
NTP client not enabled.
admin@infix-00-00-00:/> 
admin@infix-00-00-00:~$ /bin/ntp 
NTP client not enabled.
```
And again if logged in as `root` user, the command works fine:
```
admin@infix-00-00-00:~$ sudo clish
root@infix-00-00-00:/> show ntp 
Reference ID    : C23ACD94 (svl2.ntp.netnod.se)
Stratum         : 2
Ref time (UTC)  : Thu Oct 17 15:21:57 2024
System time     : 0.000035142 seconds fast of NTP time
Last offset     : -0.000068587 seconds
RMS offset      : 0.000068587 seconds
Frequency       : 0.434 ppm fast
Residual freq   : -0.011 ppm
Skew            : 28.615 ppm
Root delay      : 0.013573844 seconds
Root dispersion : 0.001952836 seconds
Update interval : 64.4 seconds
Leap status     : Normal
```

axkar, 2024-10-17:
The message "NTP client not enabled" is kind of misleading...

troglobit, 2024-10-17:
Please update infix.xml with a 'doas ' prefix for the backend shell command to the 'show ntp' CLI command. Similar to what has already been done for the container.xml. See src/klish-plugim-infix/xml/.

troglobit, 2024-10-18:
When logged in to Bash you can use `sudo chronyc serverstats`, but from src/klish-plugin-infix/xml/infix.xml, in the CLI, we must use the `doas` instead of `sudo`.   The `doas` is a wrapper I made for sudo to prevent sudo from spewing out unwanted error messages for unauthenticated users (e.g, a guest account without administrator privileges).

commit 3fa820b4 (2024-10-18): klish-plugin-infix: allow admin to run 'show ntp'

Fixes #730
closed 2024-10-18 by axkar