kernelkit/infix #777: Authorized SSH key not applied to startup config

getabc, 2024-10-28:
### Current Behavior

I can't save the authorized SSH key.

**Config applied**
```
admin@r2s:/config/> edit system authentication user admin
admin@r2s:/config/system/authentication/user/admin/> set authorized-key controller 
admin@r2s:/config/system/authentication/user/admin/> set authorized-key controller algorithm ssh-rsa
admin@r2s:/config/system/authentication/user/admin/> set authorized-key controller key-data [key]
admin@r2s:/config/system/authentication/user/admin/> commit
admin@r2s:/config/system/authentication/user/admin/> leave

```
**Running config is correct**
```
admin@r2s:/> show running-config
...
  "ietf-system:system": {
    "hostname": "r2s-1",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "authorized-key": [
            {
              "name": "controller",
              "algorithm": "ssh-rsa",
              "key-data": [key]
            }
          ],
          "infix-system:shell": "bash"
        }
      ]
```
**Config not being applied to startup config**
```
admin@r2s:/> copy running-config startup-config 
admin@r2s:/> show startup-config 
...
"ietf-system:system": {
    "hostname": "r2s",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "infix-system:shell": "bash"
        }
      ]
    },
...
```
**Workaround**
Setting the hostname saves the SSH key
```
admin@r2s:/config/> set system hostname r2s-1
admin@r2s:/config/> commit
admin@r2s-1:/config/> exit
admin@r2s-1:/> copy running-config startup-config 
...
  "ietf-system:system": {
    "hostname": "r2s-1",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "authorized-key": [
            {
              "name": "controller",
              "algorithm": "ssh-rsa",
              "key-data": [key]
            }
          ],
          "infix-system:shell": "bash"
        }
      ]

```

### Expected Behavior

```
admin@r2s:/> show startup-config
...
  "ietf-system:system": {
    "hostname": "r2s-1",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "authorized-key": [
            {
              "name": "controller",
              "algorithm": "ssh-rsa",
              "key-data": [key]
            }
          ],
          "infix-system:shell": "bash"
        }
      ]
```

### Steps To Reproduce

- Configure SSH key
- copy running config to startup config
**Not applied**
- set hostname
- copy running config to startup config
**SSH keys applied**

troglobit, 2024-10-29:
Hmm which version of Infix are you using?  Because I cannot reproduce the problem on the latest build I'm on:

```
admin@r2s:/> show version 
Infix -- a Network Operating System v24.10.1-44-g1e5d4612 -- Oct 29 05:35 CET 2024
```

getabc, 2024-10-29:
It's not the latest; I'll upgrade after the release of the OpenSSL R2S update.

```
admin@r2s-1:/> show version 
Infix -- a Network Operating System v24.10.1-23-g8fc0ab9b -- Oct 23 17:07 PDT 2024
admin@r2s-1:/> 
```

troglobit, 2024-10-29:
> It's not the latest; I'll upgrade after the release of the OpenSSL R2S update.

OK, it's been merged now.

> ```
> admin@r2s-1:/> show version 
> Infix -- a Network Operating System v24.10.1-23-g8fc0ab9b -- Oct 23 17:07 PDT 2024
> admin@r2s-1:/> 
> ```

Huh, that's a bit worrying ... well, make sure to do a full rebuild of the `r2s_defconfig` from distclean so you don't get any weird transients from an incremental rebuild.

getabc, 2024-10-29:
I have a similar result running the latest
```
admin@r2s:/> show version 
Infix -- a Network Operating System v24.10.1-49-g3e8ead6f -- Oct 29 10:42 PDT 2024
```
Factory the device - WAN set the DHCP at default, so able to access the WAN address

**Commands**
```
edit system authentication user admin
set authorized-key controller 
set authorized-key controller algorithm ssh-rsa
set authorized-key controller key-data [key]
commit
leave
copy running-config startup-config 
show startup-config 
...
  "ietf-system:system": {
    "hostname": "r2s",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "infix-system:shell": "bash"
        }
      ]
    },
```
**Workaround**
```
set system hostname r2s-1
commit
leave
copy running-config startup-config 
show startup-config 
...
{
  "ietf-system:system": {
    "hostname": "r2s-1",
    },
    "authentication": {
      "user": [
        {
          "name": "admin",
          "password": "$factory$",
          "authorized-key": [
            {
              "name": "controller",
              "algorithm": "ssh-rsa",
              "key-data": [key]            }
          ],
          "infix-system:shell": "bash"
        }
      ]
    },

```

troglobit, 2024-10-30:
This is super weird, I just rebuilt from distclean, upgraded and did a factory reset.

```
admin@r2s:/> show version 
Infix -- a Network Operating System v24.10.1-49-g3e8ead6f -- Oct 30 06:07 CET 2024
```

Then copy-pasted your commands (including `commit` which I never use) ... still works fine for me.  The authorized-key both in `running-config` and `startup-config` after saving it.

Sorry, unless you can find something in the logs (`/log/syslog`), I just cannot replicate the problem.

mattiaswal, 2025-01-30:
I can confirm this bug on latest main:
1. set key
2. copy running startup
3. nothing in startup
4. set hostname
5. the key is saved to startup

commit a64c9695 (2025-01-30): Fix issue with storing authorized SSH key to startup

The problem was that the sysrepo callback was never called,
except on update (not availible on startup)

This fix #777
PR #920 (2025-01-30): Fix issue with storing authorized SSH key to startup
The problem was that the sysrepo callback was never called, except on update (not availible on startup)

This fix #777

## Description

## Checklist

Tick *relevant* boxes, this PR is-a or has-a:

- [ ] Bugfix
  - [ ] Regression tests
  - [ ] ChangeLog updates (for next release)
- [ ] Feature
  - [ ] YANG model change => revision updated?
  - [ ] Regression tests added?
  - [ ] ChangeLog updates (for next release)
  - [ ] Documentation added?
- [ ] Test changes
  - [ ] Checked in changed Readme.adoc (make test-spec)
  - [ ] Added new test to [... cut]
closed 2025-01-30 by mattiaswal