kernelkit/infix #807: OSPF: Router id sometimes is not selected correctly

mattiaswal, 2024-11-06:
### Current Behavior

When apply a whole configuration, the router id (sometimes) gets selected before all interfaces/address is created
```
admin@R1:/> show ospf neighbor 

Neighbor ID     Pri State           Up Time         Dead Time Address         Interface                        RXmtL RqstL DBsmL
192.168.4.1       1 Full/-          22m17s             3.866s 10.0.41.1       e5:10.0.41.2                         0     0     0
192.168.3.1       1 Full/DR         22m14s             3.479s 10.0.13.2       e6:10.0.13.1                         0     0     0
1.1.1.1           1 Full/Backup     22m10s             3.200s 10.0.12.2       e7:10.0.12.1                         0     0     0
```

### Expected Behavior

```
admin@R1:/> show ospf neighbor 

Neighbor ID     Pri State           Up Time         Dead Time Address         Interface                        RXmtL RqstL DBsmL
10.0.0.4         1 Full/-          22m17s             3.866s 10.0.41.1       e5:10.0.41.2                         0     0     0
10.0.0.3         1 Full/DR         22m14s             3.479s 10.0.13.2       e6:10.0.13.1                         0     0     0
1.1.1.1          1 Full/Backup     22m10s             3.200s 10.0.12.2       e7:10.0.12.1                         0     0     0
```

troglobit, 2024-11-12:
CCB (core team meeting): should be more stable from a reboot since interfaces are created and set up at boot, in runlevel S, and ospfd is started in runlevel 2. Decision, target: future.

mattiaswal, 2024-11-25:
Fixed in #820

closed 2024-11-25 by mattiaswal