kernelkit/infix #815: YANG if-feature does not work in a sub-module if the feature is declared in the module.

getabc, 2024-11-08:
### Current Behavior

We are currently testing infix against two systems, the clixon-controller and Yangsuite

https://github.com/clicon/clixon-controller
https://github.com/CiscoDevNet/yangsuite/

### clixon-controller 
During connection, the controller reported the following error.
`Yang module infix-if-container has IF_FEATURE containers, but no such FEATURE statement exists.`

The "if-feature" does not seem to work in a sub-module if the feature is declared in the module to which it belongs.

According to RFC 7950 (YANG 1.1), Section 5.1: "A submodule can reference any definition in the module it belongs to and in all submodules included by the module." However, this does not explicitly mention the inheritance or referencing of features across modules and submodules.

### Yangsuite

I used the Cisco Yangsuite to validate all the models, and it had a similar error
The following modules in the YANG set reported errors and warnings when parsed and validated.

```
    infix-containers @ 2024-10-14 
        WARNING line 6: imported module "ietf-yang-types" not used
    infix-hardware @ 2024-04-25 
        WARNING line 9: imported module "iana-hardware" not used
    infix-if-bridge @ 2024-08-26 
        WARNING line 7: imported module "ietf-yang-types" not used
        WARNING line 19: imported module "ietf-inet-types" not used
      infix-if-container @ 2024-01-15 
        CRITICAL ERROR line 54: feature "containers" not found in module "infix-if-container"**
        WARNING line 16: imported module "infix-if-type" not used
        WARNING line 81: default value for a key leaf is ignored
    infix-ip @ 2024-09-16 
        WARNING line 15: imported module "ietf-yang-types" not used
    infix-routing @ 2024-10-01 
        CRITICAL ERROR line 455: node ietf-ospf::database is not found
        CRITICAL ERROR line 458: node ietf-ospf::database is not found
        CRITICAL ERROR line 461: node ietf-ospf::database is not found
        CRITICAL ERROR line 464: node ietf-ospf::database is not found
        CRITICAL ERROR line 467: node ietf-ospf::database is not found
        CRITICAL ERROR line 470: node ietf-ospf::database is not found
        WARNING line 18: imported module "ietf-interfaces" not used
        WARNING line 21: imported module "ietf-routing-types" not used
    infix-syslog @ 2024-07-19 
        WARNING line 148: default value for a key leaf is ignored
    libnetconf2-netconf-server @ 2024-07-09 
        WARNING line 24: imported module "ietf-tls-server" not used
```

[image]

### Conclusion
Should a submodule be able to reference any definition in the module to which it belongs?

The RFC seems unclear, and using a Cisco tool throws an error at it.

### Expected Behavior

NA

troglobit, 2024-11-15:
Reproduced with `pyang`, all of the critical errors.  I have a patch in my local tree that seems to make `pyang` happy, so I'll test with that before pushing a fix.

troglobit, 2024-11-16:
@getabc the [`container-fixes`](https://github.com/kernelkit/infix/tree/container-fixes) branch aims to fix this issue. Provided all internal testing pass it will be merged and released as v24.11.0 this coming Wednesday.

troglobit, 2024-11-20:
Fixed in 43635c2, verified by @getabc over Discord.

closed 2024-11-20 by troglobit