kernelkit/infix #855: User admin sometimes fails to be added to 'wheel' group

mattiaswal, 2024-12-04:
Most likely this is caused by all other groups are handled with change_auth but wheel is handled by change_nacm (int ietf-system), and sometimes when the admin user should be added to wheel, the user admin does not exist.

To reproduce this:
run the test: ietf_system/upgrade in a loop on virtual devices.

troglobit, 2024-12-05:
CCB: Blocker for v25.01

mattiaswal, 2025-01-29:
Unable to reproduce, ran the test ietf_system/user_admin which specifically test for this 12000 times. Occurred 0 times.

mattiaswal, 2025-07-29:
Reproduced it now:
root@infix-00-00-00:~$ groups  admin
admin sys-cli
root@infix-00-00-00:~$ 

Very rare, but we need to do something about the user/nacm create issue now.

troglobit, 2025-07-30:
Agreed, it's time. 

We've discussed it before AFK, the only real fix is to rewrite confd as a single threaded application. Cutting the ties with sysrepo-plugind.