kernelkit/infix #886: The Zeex subhook upstream is gone

getabc, 2025-01-13:
### Current Behavior

Currently using buildroot 2024.02.9

Package edk2-stable202308 
Unfortunately, it appears that the upstream project subhook is now gone, but is still being used as a submodule:

.gitmodules
[submodule "UnitTestFrameworkPkg/Library/SubhookLib/subhook"]
	path = UnitTestFrameworkPkg/Library/SubhookLib/subhook
	url = https://github.com/Zeex/subhook.git

### Expected Behavior

Latest releases pointing to own copy
https://github.com/tianocore/edk2-subhook

mattiaswal, 2025-01-13:
I have no problem to build Infix x86_64 after doing this:
<pre>
rm output/build/edk2-* -rf
rm dl/edk2* -rf
</pre>

Are you use any custom configuration? The issue you are referring seems to be fixed on buildroot master (and will be included in the next LTS of buildroot). 

https://lore.kernel.org/buildroot/20250104214053.E4E17839E1@busybox.osuosl.org/T/

mattiaswal, 2025-01-25:
This will not be fixed before upgrading buildroot to 2025.02.

closed 2025-01-25 by mattiaswal