kernelkit/infix #892: Worrying rousette warnings in log

troglobit, 2025-01-15:
Suggest patching these two logs that might look a bit worrying to end users:

```
Jan 14 05:42:12 ix-00-00-00 rousette[3439]: [2025-01-14 05:42:12.277] [rousette] [info] NACM config validation: Anonymous user access disabled 
Jan 14 05:42:12 ix-00-00-00 rousette[3439]: [2025-01-14 05:42:12.278] [rousette] [warning] Telemetry disabled. No CzechLight YANG modules found. 
```

mattiaswal, 2025-01-20:
Agree!

mattiaswal, 2025-01-21:
I would say that the first "info" is more of a security issue that needs to be patched away in rousette (and upstreamed):
From the Readme of rousette, annonymous access is enabled when:
```
1. The first entry of `rule-list` list must be configured for `ANONYMOUS_USER_GROUP`.
2. All the rules except the last one in this rule-list entry must enable only "read" access operation.
3. The last rule in the first rule-set must be a wildcard rule that disables all operations over all modules.

The anonymous user access is disabled whenever these rules are not met.

```

mattiaswal, 2025-01-21:
For anonymous we should add a switch to rousette that prefered enable anonymous mode, and test if they accept it.