kernelkit/infix #923: Let implicit default route for IPv4 autoconf be configurable

jovatn, 2025-01-31:
### Description

## Suggestion

It would be good to be able to disable the implicit default route which Infix currently sets up on interfaces with IPv4 autoconfiguration.
_Perhaps_ by an additional `no default-route` setting in the autoconf context 

```
admin@infix-00-00-00:/config/interface/e1/ipv4/autoconf/> help
  enabled                           Use a ZeroConf/IPv4LL agent to retrieve an 169.254/16 address.
  request-address                   Try to acquire the specified IP address, if available.
admin@infix-00-00-00:/config/interface/e1/ipv4/autoconf/> 
```

## Motivation

When configuring IPv4 autoconfiguration (169.254.x.x/16, also known as zeroconf) for an interface, Infix implicitly adds a default route through that interface. This default route has lowest possible routing PREFerence (254), so any default route learnt elsewhere will override.

The purpose of this default route is to facilitate easy setup with autoconf, even on links where there are units with only routeable addresses. See man page of [avahi-autoipd](https://www.systutorials.com/docs/linux/man/8-avahi-autoipd/ ) for the following motivation/example of added default route
```
To allow communication between hosts that have only an IPv4LL address assigned and hosts that only
have a routable IP address assigned you may add the following routes to both network configurations:

route add -net 169.254.0.0 netmask 255.255.0.0 dev eth0 metric 99

route add default dev eth0 metric 99
```

Although this implicit default route seems like a good default, it would be good if it can be disabled.
In OSPF, you can make a router inject a default route into OSPF, either `always` or only when the router itself has an active default route.
The latter is done by `set default-route-advertise enabled` in OSPF context, and is good to use to avoid routing to a dead end.
This OSPF use case would be incompatible with use of IPv4 autoconfiguration on the same router, unless we can disable the implicit default route.

### Additional Information

A suitable test case would be 
- to let router A inject default route info OSPF, while having some interface with IPv4 autoconfig enabled
- to let router B see if it gets a default route via OSPF from router A

Without disabling the implicit default route, B would get a default route from A via OSPF.
If A disables the implicit default route, it will no longer advertise it and B would not get a default route.

### General Information

Anyone can help out by [sponsoring][1] development of new features or [contributing][2] pull requests.
Please use this issue for discussions related to the feature.

[1]: https://github.com/kernelkit/infix/blob/main/.github/SUPPORT.md
[2]: https://github.com/kernelkit/infix/blob/main/.github/CONTRIBUTING.md

troglobit, 2025-02-03:
Alternative, hard-coded solution, for Infix -- route maps!

```
route-map filter-defaults deny 10
  match distance 254
  match ip address prefix-list def-route
!
route-map filter-defaults permit 20
  match ip address prefix-list def-route
!
ip prefix-list def-route permit 0.0.0.0/0
```

I have not tried it, but with this we could deny all routes that have distance==254 from being redistributed.

troglobit, 2025-02-07:
CCB: decsion, drop default route entirely!  No need for "automatic" or clever handling, user can set this up manually, if needed.

PR #957 (2025-02-24): Misc
## Description

 - Fix #923: drop default route from IPv4 autoconf
 - Fix #956: don't `chown()` on copy targets like `running-config`

## Checklist

Tick *relevant* boxes, this PR is-a or has-a:

- [X] Bugfix
  - [ ] Regression tests
  - [x] ChangeLog updates (for next release)
- [ ] Feature
  - [ ] YANG model change => revision updated?
  - [ ] Regression tests added?
  - [ ] ChangeLog updates (for next release)
  - [ ] Documentation added?
- [ ] Test changes
  - [ ] Checked in changed Readme.adoc (make test-spec)
  - [ ] Added new test to group Readme.adoc and yaml file
- [ ] Code style [... cut]
commit f690cd21 (2025-02-24): Fix #923: drop default route for IPv4 autoconf
closed 2025-02-24 by troglobit