kernelkit/infix #930: Container reconfiguration is broken

wkz, 2025-02-05:
### Current Behavior

After a container has been launched any further reconfiguration is not applied until after reboot or a manual `container {stop, setup, start}` cycle.

### Expected Behavior

1. Setup `docker://nginx:alpine`
2. Activate

As expected, the default landing page is accessible

3. Add a content mount with a custom landing page
4. Activate

The custom landing page is expected to be seen, but the default is still shown

### Additional information

The root cause is that the download/setup phase for the container is run by `finit` as a `task`. So unless the runlevel changes (which it does not during a normal reconfig-cycle), the setup phase is not run again.

troglobit, 2025-02-07:
- Related to #902
- How the container setup task is run needs to be refactored, possibly as a finit `pre:` script
- Update basic test to verify changing configuration of an existing container works

jovatn, 2025-02-10:
@troglobit  Milestone 25.02?

troglobit, 2025-02-10:
> @troglobit  Milestone 25.02? I will assume this was the intention during last CCB, and add it to 25.02 now.

Indeed, thanks for keeping track 🙏

commit 8e8ad921 (2025-02-23): container: refactor setup stage to use Finit pre: script

When updating a container configuration we need to rerun the setup stage
of the container.  However, a run/task in Finit only runs once for each
runlevel, meaning changes at runtime of a container was broken.

This is a refactor of the container script to allow it to also run as a
pre:script in the Finit setup stage of a sysv/service.  Also included is
a fix to the timeout/retry handling in case the container setup fails.
We must call `read -t 60` *before* calling `ip monitor`, otherwise it
will not be called until `ip monitor` returns. [... cut]
PR #955 (2025-02-25): Lingering containers
## Description

 - Bump Finit for another set of bug fixes, and `cleanup:script` support
 - Fix #902 by relocating removal from `confd` to Finit sysv `cleanup:script`
 - Fix #930 by switching from a Finit `task` (runs once per runlevel) to a `pre:script`

Also, a new container test added, to verify volumes are not lost on container upgrade.  However, upgrading is only done to the same OCI image that the container was initially created from.

As an added bonus, the `container` helper script now supports running commands inside a running Docker container.  This works even ov [... cut]
commit 3811df9a (2025-02-25): container: refactor setup stage to use Finit pre: script

When updating a container configuration we need to rerun the setup stage
of the container.  However, a run/task in Finit only runs once for each
runlevel, meaning changes at runtime of a container was broken.

This is a refactor of the container script to allow it to also run as a
pre:script in the Finit setup stage of a sysv/service.  Also included is
a fix to the timeout/retry handling in case the container setup fails.
We must call `read -t 60` *before* calling `ip monitor`, otherwise it
will not be called until `ip monitor` returns. [... cut]