kernelkit/infix #932: Non-privileged users do not have access to sysrepocfg

wkz, 2025-02-07:
### Current Behavior

When granting a non-privileged user shell access via `clish`, many admin/exec commands are broken.

Using {NET,REST}CONF though, the user is able to access the expected data.

### Expected Behavior

`clish` should have access to the same data set that is available to the user via {NET,REST}CONF.

### Steps To Reproduce

1. Create a non-privileged user, "guest", with `clish` as their shell
2. Login as "guest"
3. Run `show interfaces` or `show software`, for example

### Additional information

The root cause seems to be that the user can not extract any data via `sysrepocfg`, which `clish` uses for most of its admin/exec commands.

troglobit, 2025-02-07:
CCB: a guest user should be able to access sysrepo, but be blocked access by the NACM layer (in sysrepo) from what they can access.

Add new UNIX group, e.g. `sysrepo-users`, which the `/dev/shm/*sysrepo*` files are read-write group member of.  This way even a guest user will be member of, meaning they will be able to connect to sysrepo to query status -- filtered via NACM rules.

We also need to add `admin` users to the `frrvtysh` group (like we already do for the `wheel` (root) group).

mattiaswal, 2025-04-04:
This may be the solution we are looking for:

[image]

commit 7527135f (2025-04-28): sysrepo: allow non-wheel users access to sysrepo data

Non-privileged users in the sys-cli group can now access sysrepo data.

The data is now solely protected by the NCAM rules and not Unix file
permissions. Any stray user that's not part of the (default) sys-cli
group still can't access syrepo at all, like users added from the shell.

Fixes #932
commit ac89f945 (2025-04-28): sysrepo: allow non-wheel users access to sysrepo data

Non-privileged users in the sys-cli group can now access sysrepo data.

The data is now solely protected by the NCAM rules and not Unix file
permissions. Any stray user that's not part of the (default) sys-cli
group still can't access syrepo, like users added from the shell.

Fixes #932
closed 2025-04-29 by rical