kernelkit/infix #939: Add LLDP information to statd-cli

axkar, 2025-02-10:
### Description

Now that statd supports LLDP and supplies LLDP data to sysrepo, the statd-cli can now include a show-lldp command.

### General Information

Anyone can help out by [sponsoring][1] development of new features or [contributing][2] pull requests.
Please use this issue for discussions related to the feature.

[1]: https://github.com/kernelkit/infix/blob/main/.github/SUPPORT.md
[2]: https://github.com/kernelkit/infix/blob/main/.github/CONTRIBUTING.md

jovatn, 2025-04-16:
When fixing this issue, can you also provide example of this AND the ability to make LLDP settings per port under https://github.com/kernelkit/infix/blob/main/doc/discovery.md#lldp ?
And make a statement in Changelog on LLDP per port settings?

axkar, 2025-04-16:
> When fixing this issue, can you also provide example of this AND the ability to make LLDP settings per port under https://github.com/kernelkit/infix/blob/main/doc/discovery.md#lldp ? And make a statement in Changelog on LLDP per port settings?

I'll do that :)