kernelkit/infix #1052: Discuss, new slogan?

troglobit, 2025-05-25:
As Infix is maturing, and being ported to more and more devices, it is outgrowing its use a a pure Network Operating System.  Some new devices are switches/routers, but some, like the Raspberry Pi, are more of end devices.

So, I'd like to bring up a discussion around the current slogan: *"Infix -- a Network Operating System"* to something that invites more devices and users.  Here are a few I came up with during the weekend:

 - Infix — Secure by Design
 - Infix — Immutable Embedded OS
 - Infix — Robust. Immutable. Secure.
 - Infix — Embedded OS, Redefined
 - Infix — Embedded Reliability
 - Infix — Always Reliable
 - Infix — Built for Security

wkz, 2025-05-26:
Infix --- Linux ❤ YANG

jovatn, 2025-05-26:
I like all the above. 
Anyone who can come up with something explicitly hinting about the "automation" aspect?
Linux ❤ YANG is very informative for those who know YANG

What I think of is something inline with the following, but more catchy
Infix -- Designed for automation

troglobit, 2025-06-01:
[image]

jovatn, 2025-06-16:
"Immutable. Friendly. Secure." it is!
Update info on github.

closed 2025-06-24 by troglobit